ros-controls/ros2_controllers #432: [JointStateBroadcaster] Add tests for clearing joint_names in JointStateBroadcaster

destogl, 2022-09-06:
Two cases should be tested:

1. When broadcaster is reactivated, joints should not be duplicated.
2. When broadcaster is deactivated, the joint names should be cleaned.

That's technically be more correct, but I think it's even more bulletproof to let the function generating a new setup do a quick cleanup -- as-is now

_Originally posted by @bmagyar in https://github.com/ros-controls/ros2_controllers/pull/431#discussion_r963292043_

christophfroehlich, 2025-03-14:
@saikishor you fixed that, and we already have a test for that right?

saikishor, 2025-03-20:
Hello!
@christophfroehlich Yes you are right. This is already handled in https://github.com/ros-controls/ros2_controllers/pull/1331 :)

closed 2025-03-20 by christophfroehlich