ros-controls/ros2_controllers #440: Disable cmd_vel_timeout

matthew-william-lock, 2022-09-29:
Hey !

Is there any way to disable the cmd_vel_timeout for the diff drive controller? It is not very apparent how to do this in the documentation. I have also checked the source code as well and it does not seem possible in its current implementation. 

I am happy to create a PR if needed.

See this https://github.com/cyberbotics/webots_ros2/issues/468 for more context.

DrAndyWest, 2023-06-27:
I just want to echo the need for this enhancement.  Had awful trouble with Clearpath Jackal hardware, where the [twist_mux](https://github.com/ros-teleop/twist_mux) uses it's own timeout, and does not repeat joy commands if they don't change.

Therefore, the robot would suddenly stop even when commanding full velocity if the joystick input wasn't changing.  Work around was to add a large timeout for the diff drive controller.

A `0.0 = ignore` option like most ROS nodes would be perfect.

gavanderhoorn, 2023-06-27:
> and does not repeat joy commands if they don't change.

isn't that a `joy` (as in: the node) or even an OS-level setting? Afaik, `twist_mux` doesn't even interpret the incoming msgs. It [just forwards them](https://github.com/ros-teleop/twist_mux/blob/1b9cc358e1d607da1d0678374d3890eefe2d0d58/include/twist_mux/topic_handle.hpp#L187-L199).

DrAndyWest, 2023-06-27:
I found the behaviour also happened for simple command-line publishing at a set rate (avoiding the autorepeat in the joy node), into twist_mux.  I am using an autorepeat_rate of 20 Hz, therefore, the default.

Twist Mux does look to check if there has been a [change in values](https://github.com/ros-teleop/twist_mux/blob/c6d4926b7bcf80a66d5118445bce33c84d431180/src/twist_mux.cpp#L53), but I am not sure this actually gets used - your link to the callback being evidence it is not.

Either way, I do feel a bit uncomfortable with the controller being able to issue zero velocity commands when the command input is non-zero, from a transparency point of view.

gavanderhoorn, 2023-06-27:
> Twist Mux does look to check if there has been a [change in values](https://github.com/ros-teleop/twist_mux/blob/c6d4926b7bcf80a66d5118445bce33c84d431180/src/twist_mux.cpp#L53), but I am not sure this actually gets used - your link to the callback being evidence it is not.

looks like that's [dead code indeed](https://github.com/search?q=repo%3Aros-teleop%2Ftwist_mux%20hasIncreasedAbsVelocity&type=code).

> Either way, I do feel a bit uncomfortable with the controller being able to issue zero velocity commands when the command input is non-zero, from a transparency point of view.

it's pretty important functionality actually. Many a run-away robot has been saved when it lost connection to its controller for whatever reason and was driving max-velocity into a wall, or water, or something else. Standard functionality as well for velocity-control interfaces/controllers.

But again: not claiming it can't be better documented or should be something that can be disabled.

mwlock, 2025-04-11:
Bumping the issue to keep alive for further comment/consideration 😄

christophfroehlich, 2025-05-21:
Setting the cmd_vel_timeout parameter to 0.0 fixes this, right? However, currently it is not updated in active state of the controller, but is this a must-have? (should either be changed or the parameter declared as read-only)

DrAndyWest, 2025-07-16:
@christophfroehlich - it looks like the check `cmd_vel_timeout_ == rclcpp::Duration::from_seconds(0.0)` is performed on configure for the current master branch ([see here ](https://github.com/ros-controls/ros2_controllers/blob/a10c56463f8ccf54b9ad0405aa3f9b1285ddb6a2/diff_drive_controller/src/diff_drive_controller.cpp#L393)), but not in the actual controller update function ([master branch](https://github.com/ros-controls/ros2_controllers/blob/a10c56463f8ccf54b9ad0405aa3f9b1285ddb6a2/diff_drive_controller/src/diff_drive_controller.cpp#L118), [humble branch](https://github.com/ros-controls/ros2_controllers/blob/d692710195501ca6e30c0ba796833e9363c86094/diff_drive_controller/src/diff_drive_controller.cpp#L126)).  Therefore, setting the parameter to 0.0 will still trigger the braking behaviour, as `age_of_last_command` is always greater than zero.

When you say

> However, currently it is not updated in active state of the controller, but is this a must-have?

Do you mean that it is not "dynamic"?  I _personally_ think it should be read-only parameter, I don't see a need to change the timeout value on the fly, given all the previous comments around autorepeat rates and such.

closed 2025-10-16 by github-actions[bot]