ros-controls/ros2_controllers #457: Chained Controller exported reference interface unavailable

Teusner, 2022-10-28:
## Bug
When we implement a chained controller, we need to export reference interfaces which will be (as far as I understood) seen by other controllers as command interfaces. The problem is that these exported reference interfaces don't seem to be available (as they appear unavailable). Here is the listing of the hardware interfaces.

```bash
> ros2 control list_hardware_interfaces
command interfaces
        joint1/position [available] [claimed]
        controller1/position [unavailable] [unclaimed]
[...]
```
It induces that the previous controller can't find the exported reference interface, configured as a command interface in this one, and this previous controller can't be properly initialized.

I may have miss something to make these exported reference interfaces available, it will be more clear when the documentation will be available :kissing_heart:. In my controller I had to add the following instructions to export my reference interfaces : 

```c++
std::vector<hardware_interface::CommandInterface> TFBroadcaster::on_export_reference_interfaces() {
        std::vector<hardware_interface::CommandInterface> command_interfaces;
        command_interfaces.emplace_back(hardware_interface::CommandInterface(get_node()->get_name(), "position", &reference_interfaces_[0]));
        return command_interfaces;
}
[...]
controller_interface::CallbackReturn TFBroadcaster::on_configure(const rclcpp_lifecycle::State & /*previous_state*/) {
        [...]
        reference_interfaces_.resize(1, std::numeric_limits<double>::quiet_NaN());
        [...]
}
[...]
controller_interface::CallbackReturn TFBroadcaster::on_activate(const rclcpp_lifecycle::State & /*previous_state*/) {
        std::fill(reference_interfaces_.begin(), reference_interfaces_.end(), std::numeric_limits<double>::quiet_NaN());
        return CallbackReturn::SUCCESS;
}
```

## Configuration
- OS: Ubuntu 22.04
- Version: ROS2 Humble
- ROS2Control: installed from apt binary packages (http://packages.ros.org/ros2/ubuntu jammy/main amd64)

Teusner, 2022-11-07:
**Update** It seems that it could be a `ros2cli` issues but I'm not sure...

```bash
> ros2 control list_controllers 
controller1     [controllers/Controller] unconfigured
```

```bash
ros2 control set_controller_state controller1 configure
usage: ros2 control set_controller_state [-h] [--spin-time SPIN_TIME] [-s] [-c CONTROLLER_MANAGER] [--include-hidden-nodes] controller_name {inactive,active}
ros2 control set_controller_state: error: argument state: invalid choice: 'configure' (choose from 'inactive', 'active')
```
(don't know if this has to be `configure` or `configured`, tried both, same output :sob: )

Can you confirm that exported reference interfaces are unavailable because my controller is unconfigured ? How can I then configure my controller (Updating the CLI, using a service call, ...) ?

Thanks in advance

maxwelllls, 2024-01-20:
After using the admittance_controller, the controller interface changes to the following:

> boomy-controller:~$ ros2 control list_hardware_interfaces
> command interfaces
>     joint1/position [available] [claimed]
>     joint2/position [available] [claimed]
>     joint3/position [available] [claimed]
>     joint4/position [available] [claimed]
>     joint5/position [available] [claimed]
>     joint6/position [available] [claimed]
>     rml63_admittance_controller/joint1/position [available] [claimed]
>     rml63_admittance_controller/joint1/velocity [available] [unclaimed]
>     rml63_admittance_controller/joint2/position [available] [claimed]
>     rml63_admittance_controller/joint2/velocity [available] [unclaimed]
>     rml63_admittance_controller/joint3/position [available] [claimed]
>     rml63_admittance_controller/joint3/velocity [available] [unclaimed]
>     rml63_admittance_controller/joint4/position [available] [claimed]
>     rml63_admittance_controller/joint4/velocity [available] [unclaimed]
>     rml63_admittance_controller/joint5/position [available] [claimed]
>     rml63_admittance_controller/joint5/velocity [available] [unclaimed]
>     rml63_admittance_controller/joint6/position [available] [claimed]
>     rml63_admittance_controller/joint6/velocity [available] [unclaimed]

As a result, the "controller" combobox widget in rqt_joint_trajectory_controller cannot list the current controller.

The following modification to joint_trajectory_controller.py allows it to correctly list the controller. 

This code by default gets the first parameter of the interface. However, after chained, a prefix is added to the command interfaces, so here it is modified. The last one parameter is usually position/velocity/effort, and the second-to-last parameter is the joint name.
```

def _jtc_joint_names(jtc_info):
    # NOTE: We assume that there is at least one hardware interface that
    # claims resources (there should be), and the resource list is fetched
    # from the first available interface

    joint_names = []
    for interface in jtc_info.claimed_interfaces:
        # name = interface.split("/")[0]
        name = interface.split("/")[-2]
        joint_names.append(name)

    return joint_names
```
I will be submitting a pull request for this."

christophfroehlich, 2025-02-11:
I hope this issue is fixed with the demo we provide on controller chaining https://control.ros.org/humble/doc/ros2_control_demos/example_12/doc/userdoc.html

closed 2025-02-11 by christophfroehlich