ros-controls/ros2_controllers #458: Create a Humble branch that builds.

boilerbots, 2022-10-31:
Also... Are you building from source on humble or on rolling? The master branches target rolling and may only work with the latest distro up to a point... There's a specific branch for humble.

_Originally posted by @bmagyar in https://github.com/ros-controls/ros2_controllers/issues/454#issuecomment-1296335979_

Please create a humble branch before the changes that where introduced into master that break the build for Humble.

jmguerreroh, 2022-11-02:
Using this branch, you can build your project in Humble: https://github.com/jmguerreroh/ros2_controllers.git

bmagyar, 2022-11-02:
The build for admittance_controller should be fixed after the next sync.

bmagyar, 2022-11-27:
Closing due to no activity

closed 2022-11-27 by bmagyar