ros-controls/ros2_controllers #462: Test all JTC parameters to be set properly from YAML file when using `generate_parameters_library`

destogl, 2022-11-14:
## Background

Add all parameters from JTC into [the example/test YAML file](https://github.com/ros-controls/ros2_controllers/blob/master/joint_trajectory_controller/test/config/test_joint_trajectory_controller.yaml).

To add all parameters check the definition of parameters in [this YAML file](https://github.com/ros-controls/ros2_controllers/blob/master/joint_trajectory_controller/src/joint_trajectory_controller_parameters.yaml) and add them with non-default values.

Also, don't forget to adjust test to still work correctly.

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_controllers/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/foxy/Tutorials/Workspace/Creating-A-Workspace.html), for Step3 use "Download Source Code" section with [these instructions](https://ros-controls.github.io/control.ros.org/getting_started.html#compiling).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_controllers`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_controllers` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://ros-controls.github.io/control.ros.org/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/foxy/Tutorials.html)
* [ROS Answers](https://answers.ros.org/questions/)

**Good luck with your first issue!**

RamCharanThota, 2022-11-22:
Hi @destogl, My name is Ram. I have a some decent experience with ROS. I have been looking for opportunities to contribute to opensource. But, I don't have any previous experience. I am interested to take this opportunity and work on the above issue. You can assign this issue to me. Thank you :)

RamCharanThota, 2022-12-19:
Hi,
I wondering whether I can start working on the task or do i need to wait until someone assign this task to me?.Sorry I am very new to open source and don't have much knowledge about it. Thank you

christophfroehlich, 2023-04-08:
> Hi, I wondering whether I can start working on the task or do i need to wait until someone assign this task to me?.Sorry I am very new to open source and don't have much knowledge about it. Thank you

Hi there. It seems that your message got lost, sorry. Are you still up to contributing? We don't necessarily assign issues, a comment here that you are working on it is sufficient as written in the issue description.

bmagyar, 2023-04-10:
Welcome and thanks!!

1) You can find the parameters and their default values here: https://github.com/ros-controls/ros2_controllers/blob/master/joint_trajectory_controller/src/joint_trajectory_controller_parameters.yaml
The description asks you to add tests with values other than the default ones to ensure we don't only test those.
2) Adjusting the parameters passed to the controller may cause some of the tests to fail. You'll have to look at those tests and try to fix them. Don't worry we are here to help. You can run tests with `colcon test --packages-select joint_trajectory_controller` and examine the output with `colcon test-results`

RamCharanThota, 2023-04-11:
Thank you for the explanation @bmagyar . I am bit confused about "command_joints" parameter in the yaml file you mentioned. They aren't any default_value for that parameter. Can I define any joint names of my own or does these default values need to be something specific. Thank you

christophfroehlich, 2023-04-11:
The [URDF file used for the test](https://github.com/ros-controls/ros2_controllers/blob/master/joint_trajectory_controller/test/test_load_joint_trajectory_controller.cpp#L35) is the [minimal_robot_urdf from here ](https://github.com/ros-controls/ros2_control/tree/humble/ros2_control_test_assets).
If you don't specify `command_joints`, the joints from `joints` parameter are used.
For example: A test here should check what happens if a non-existing joint is given for this parameter.

RamCharanThota, 2023-04-11:
Thank you @christophfroehlich for the explanation. One more thing, do I need to build the ros2_control  before running the tests using above command (colcon test --packages-select joint_trajectory_controller). If yes, where can i find the build instructions for ros2 humble distro. Thank you

RamCharanThota, 2023-04-12:
Thanks for reply @christophfroehlich. I mean to say ros2_controllers. I did the following steps
`    1) forked the ros2_controllers repo 
     2) created the ros2_ws workspace (I am using ros2 humble) 

     $ cd ~
     $ mkdir  -p ros2_control_ws/src/ 
     $ cd ros2_control_ws/src/
     $ git clone {url of my forked repo of ros2_controller}
     $ git branch <my_new_branch>
     $ git checkout -b <my_new_branch>
     $ pip3 install pre-commit && pre-commit install
     $ cd ~/ros2_control_ws
     $ colcon build `

I am getting the following error
[image]

I tried to fix the problem by installing rcpputils package. But, it is ros 1 package. I am wondering whether I missed any installing any other packages or build steps. I really appreciate if you can help me this

christophfroehlich, 2023-04-12:
have you installed and sourced rolling distribution? if yes, try to run rosdep, e.g. see https://github.com/ros-controls/ros2_control/blob/master/CONTRIBUTING.md

RamCharanThota, 2023-04-13:
Thanks @christophfroehlich . After running the rosdep, I build the ros2_controllers. I had only one error due to admittance_controller pkg as shown in here
[image]

After ignoring the admittance_controller_pkg, I can build the entire ros_controllers repo without any issues

christophfroehlich, 2023-04-13:
I guess there was a recent API break and not yet released in the ros2_control package. You can COLCON_IGNORE it as you did, or also clone the ros2_control repo into your workspace and build it from source.

RamCharanThota, 2023-04-16:
Thanks @christophfroehlich  I have completed adding the parameters to yaml file and ran the tests using the following commands
`colcon test --packages-select joint_trajectory_controller

colcon test-result
`
And there are no errors. But, I am not sure whatever I have done correct or not. So, what are the next steps, I need to submit the PR and then you will let me know whether are there any issues or not?

christophfroehlich, 2023-04-16:
You can submit a PR and we can have a look at what you added, and if there is something missing. 
If you think it is still not ready, you can submit a "Draft PR" to make that clear.

RamCharanThota, 2023-04-16:
Thank you @christophfroehlich. I have submitted the draft PR. You can have a look and let me know :)

destogl, 2023-09-20:
We close this because it is obsolete.

closed 2023-09-20 by destogl