ros-controls/ros2_controllers #463: Add statistics collection to `mock_components/GenericSystem`

AndyZe, 2022-11-15:
When running simulations, it would be nice to add statistics collection to `mock_components/GenericSystem`. For example:

- What is the average rate of incoming commands?
- What is the average latency (from message stamp to message received)?
- What is the standard deviation of time between messages?

I think it would also be nice to add a service call to trigger this data collection so it doesn't begin until the user is ready.

I don't have any immediate need for this but it would have been nice on a few projects in the past.

mcbed, 2022-11-16:
Hi Andy, 
In terms of functionality inside the framework, does it change something to collect this date in a hardware interface rather than in a dedicated controller ? We use some of what you describe here but in a [controller](https://github.com/mcbed/r2c_perf_bmark) version to test our setups what allows us to collect data also if running a real hardware and I am curious if we could expect some different results.

bmagyar, 2022-11-17:
For https://github.com/ros-controls/ros2_control/pull/806 we used an instrumented JTC and instrumented mock hardware components