ros-controls/ros2_controllers #469: JTC does not stop velocity command after goal succeeds or fails

tonynajjar, 2022-11-22:
**Describe the bug**

I'm using the JTC with a velocity command interface. After the trajectory goal succeeds, a small velocity command is still continuously sent to the hardware interface.

**To Reproduce**
setup JTC with velocity command and send a goal

**Expected behavior**

I would expect the velocity command to be reset to 0.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - OS: Ubuntu Jammy
 - Version: master

**Additional context**
Add any other context about the problem here, especially include any modifications to any ros2_controllers that relate to this issue.

tonynajjar, 2022-11-23:
I noticed that this only happens when `use_closed_loop_pid_adapter_ = true`, so when there is a only a velocity command, which is my case

christophfroehlich, 2023-03-16:
Even worse (?): I realized now that JTC with `use_closed_loop_pid_adapter_` does not stop the movement after a goal gets cancelled, but the last velocity is maintained.

christophfroehlich, 2023-03-31:
We have now two PRs addressing the hold-position at startup or after canceling a goal. I think now, that the issue of this PR has a different cause because `set_hold_position()` is not called after reaching the goal:
https://github.com/ros-controls/ros2_controllers/blob/b70a8d0f685574f7e9647f8f4b4518097d505214/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L323-L333

I tried different action goals, with velocity command interface (`use_closed_loop_pid_adapter_` is true). I could not reproduce the behavior you described if the action goal has only position entries.
But if I add velocities to the points, where the last point is nonzero, the controller does not stop after reaching the goal. (behavior is depending on the combination of `ff_velocity_scale` and feedback gains)
https://github.com/christophfroehlich/gazebo_ros2_control/blob/66eabd30851847da2143714a586ec9524d79a73a/gazebo_ros2_control_demos/examples/example_position_with_velocity.cpp#L132-L137

Might that be your case? I'm not sure what the controller should do if the last received commanded velocity is different from zero. Should we call `set_hold_position()` then?

tonynajjar, 2023-03-31:
I moved on from this and don't use JTC anymore unfortunately, I won't be able to help

christophfroehlich, 2023-04-02:
Ok. If the problem was a nonzero velocity of the trajectory point at the end, it's maybe not a bug but some missing (optional) feature to automatically set it zero.

christophfroehlich, 2023-08-15:
Fixed with #558

closed 2023-08-15 by christophfroehlich