ros-controls/ros2_controllers #472: Port velocity_controllers/joint_position_controller from ROS1

tonynajjar, 2022-11-24:
**Is your feature request related to a problem? Please describe.**
I know you probably have it on your roadmap somehwere but I didn't find it in the issues (found [similar](https://github.com/ros-controls/ros2_controllers/issues/373)) so I'm writing this feature request to formalize it and track it

I might take this on in the near future.

bmagyar, 2022-11-25:
Volunteer accepted! :D

destogl, 2022-11-30:
Wait! We maybe don't need that! :)

tonynajjar, 2022-12-01:
Thanks for the heads up, how far away is the PR from being testable? I could help test it

destogl, 2023-01-11:
The PR is testable and workable (almost in production already :D) But I didn't have a chance to finish up the tests.

muttistefano, 2023-03-01:
Dear @destogl , i'm trying to test and use your PR.
In my case, I have a joint position controller robot, and I want to use the PID controller to feed positions to the hardware interface while receiving velocity commands.
Up to now my config looks like : 

```
controller_manager:
  ros__parameters:
    update_rate: 125  # Hz

    pid:
      type: "pid_controller/PidController"

    forward_velocity_controller_pid:
      type: forward_command_controller/ForwardCommandController

pid:
  ros__parameters:
    dof_names:
      - shoulder_pan_joint
      - shoulder_lift_joint
      - elbow_joint
      - wrist_1_joint
      - wrist_2_joint
      - wrist_3_joint

    command_interface: position

    reference_and_state_interfaces:
      # - position
      - velocity

    update_rate: 125

    gains:
      shoulder_pan_joint:  {"p": 1.0, "i": 1.0, "d": 0.0, "i_clamp_min": -7.0, "i_clamp_max": 7.0, "antiwindup": true}
      shoulder_lift_joint: {"p": 1.0, "i": 1.0, "d": 0.0, "i_clamp_min": -7.0, "i_clamp_max": 7.0, "antiwindup": true}
      elbow_joint:         {"p": 1.0, "i": 1.0, "d": 0.0, "i_clamp_min": -7.0, "i_clamp_max": 7.0, "antiwindup": true}
      wrist_1_joint:       {"p": 1.0, "i": 1.0, "d": 0.0, "i_clamp_min": -7.0, "i_clamp_max": 7.0, "antiwindup": true}
      wrist_2_joint:       {"p": 1.0, "i": 1.0, "d": 0.0, "i_clamp_min": -7.0, "i_clamp_max": 7.0, "antiwindup": true}
      wrist_3_joint:       {"p": 1.0, "i": 1.0, "d": 0.0, "i_clamp_min": -7.0, "i_clamp_max": 7.0, "antiwindup": true}

# preceeding_forward_velocity_controller_pid:
#   ros__parameters:
#     joints:
#       - pid/shoulder_pan_joint
#       - pid/shoulder_lift_joint
#       - pid/elbow_joint
#       - pid/wrist_1_joint
#       - pid/wrist_2_joint
#       - pid/wrist_3_joint
#     interface_name: velocity

forward_velocity_controller_pid:
  ros__parameters:
    joints:
      - pid/shoulder_pan_joint
      - pid/shoulder_lift_joint
      - pid/elbow_joint
      - pid/wrist_1_joint
      - pid/wrist_2_joint
      - pid/wrist_3_joint
    interface_name: velocity
```
Apart from the PID values that have to be tuned, it seems like the controller still receives position reference rather than velocity.
Am I doing something wrong?

This configuration is a modified version of the one provided in your pull request.
I had to comment out the preceding controller otherwise it wont work.

Thanks