ros-controls/ros2_controllers #482: diff_drive_controller generates wrong odom and robot base frame names

lukicdarkoo, 2022-12-18:
Since the last sync (https://discourse.ros.org/t/new-packages-for-ros-2-humble-hawksbill-2022-12-16/28806) there are two issues:
1. Default frame name of the robot base frame is `odom`. This used to be `base_link` and it was a meaningful name. I don't think we should change defaults within the single ROS distro. I guess the problem is here: https://github.com/ros-controls/ros2_controllers/blob/42f6c1470508ca2480b954eead58a87e7e827434/diff_drive_controller/src/diff_drive_controller_parameter.yaml#L49
2. Namespaces are passed as prefixes to the odom and robot base frame names. I think I read somewhere that adding prefixes to frame names in multi-robot systems is a bad practice. Anyways, like in the first issue, the behavior shouldn't change within a single ROS distro.
    Even adding a leading slash to the frame names didn't make prefixes go away. My temporary solution is to use the snapshot version:
    ```bash
    wget -O /tmp/diff_drive_controller.deb http://snapshots.ros.org/humble/2022-11-23/ubuntu/pool/main/r/ros-humble-diff-drive-controller/ros-humble-diff-drive-controller_2.12.0-1jammy.20221108.202153_amd64.deb && \
    sudo apt install -y /tmp/diff_drive_controller.deb && \
    rm -f /tmp/diff_drive_controller.deb
    ``` 

It looks like the changes are made on purpose in https://github.com/ros-controls/ros2_controllers/pull/461. I am not sure whether it is a good decision, especially considering that the change breaks the API.

destogl, 2023-01-11:
@lukicdarkoo you are right. I don't know how I missed this "/" issue with frames. I think we should use "_" as separator between prefix and frame name and use it only if there is controller namespace provided.

@delihus maybe you would like to open another PR 😉

destogl, 2023-01-11:
@lukicdarkoo sorry, this change slipped through into humble unfortunately…

lukicdarkoo, 2023-01-12:
@destogl I don't think the namespace should affect frame names. So far I haven't seen any package doing it in ROS 2.

Instead, there should be a parameter, like `frame_prefix`. Or, simply not allowing prefix is also fine. A user can add prefix when defining frame names anyways.

For example in Nav2, a namespace adds a prefix to every topic, including `tf` and `tf_static`. It makes transforms isolated and no need to add prefixes to frame names.

bobbleballs, 2023-02-24:
Hi, I also agree that #461 should be reverted.
I think that having a prefix option could be a way of doing it for those who only have 1 `/tf` topic and want to add differentiation.
In my situation I use remapping to make a new `/tf` topic for each namespace as described in [this issue on Geometry2](https://github.com/ros-controls/ros2_controllers/issues/521#issuecomment-1444136319). This is also how they do it on the Nav2 project, so having nothing would be best for me!

hyunseok-yang, 2023-11-09:
hi, anyone following this problem?
I guess I'm struggling with same issue .

I applied a namsapce on the node. But /tf publishes frame id with namespace as a prefix. 

All I need is just remove the namespace on frame id for TF.

nlb196, 2024-01-03:
Didn't someone mention a config parameter would be added for the frame prefix?  My current bot setup is to namespace my nodes, topics, and the tf topics.  I do not want to prefix my frame names.  The current code is automatically prefixing my frames when the node has a namespace.  

It seems like we just need a config param for the frame  prefix, and default  it to nothing.  I though I read this was going to happen, but I can't seem to find it again.

My current fix was to clone the package and remove the prefixing code, but that's not maintainable.  I'd be glad to help if nobody else is working to fix this.

christophfroehlich, 2025-01-01:
I think this was solved with #533

closed 2025-01-01 by christophfroehlich