ros-controls/ros2_controllers #483: Missing `rsl` dependency

alsora, 2022-12-21:
**Describe the bug**

Hi, I'm using the `ros2_controllers` package as part of the Create 3 robot simulator (https://github.com/iRobotEducation/create3_sim).
In particular, I need to load the differential drive controller plugin.

Using ROS 2 Humble, it looks like the `rsl` library is now a dependecy https://index.ros.org/p/rsl/github-PickNikRobotics-RSL/#humble

However, this is not pulled in automatically when installing `ros2_controllers` and/or the differential drive plugin (either via `apt-get install` or via `rosdep`).

I temporarily worked around this issue by explicitly adding `rsl` as a dependency to my project, however I think that it should be added somewhere in this repository https://github.com/iRobotEducation/create3_sim/pull/197/commits/3282415b5da1f9246fded08b343c4bc1efedcd4e

bmagyar, 2022-12-21:
Hey Alberto,

I don't think it is healthy to redeclare dependencies that should be
transitive. I believe this was fixed on the library side already, just
needs a new sync. Optionally you may want to consider using the `humble`
branch of `ros2_controllers` which may not have this issue. (I haven't
confirmed that yet)

alsora, 2022-12-21:
Yes, that was just a workaround and the reason why I popped up here with the issue =)
I see now that you added the dependency two weeks ago in `joint_trajectory_controller/package.xml`.

We will remove the hack after the new sync is broadly available.

closed 2022-12-21 by alsora