ros-controls/ros2_controllers #488: Should `admittance_controller` also depend on `kinematics_interface_kdl` package?

sea-bass, 2023-01-04:
**Describe the bug**
I'm not sure if this is a bug or a question, which is why I didn't just submit a PR...

I just ran into an issue with `admittance_controller` because we moved from a source build of the `kinematics_interface` repo to a binaries install.

In doing so, I realize that we had previously been building both `kinematics_interface` and `kinematics_interface_kdl` packages from source, but moving to binaries the rosdeps only installed `kinematics_interface`.

Given that this is just an optional kinematics plugin (albeit the only kinematics interface that comes off the shelf), should the `admittance_controller/package.xml` add `kinematics_interface_kdl` as another dependency?

destogl, 2023-01-11:
Strictly speaking, only tests have dependency to ` kinematics_interface_kdl` and I have added this some time ago, but I am not sure which sync are you using. It can be that first release went out without it, but we fixed and re-released a few days later. Can you check the version you are using?

Is it possible that you are not installing test dependencies? In that case, you should add dependency into your scenario because there is where you specify configuration of the controller and exact plugin for the IK.

sea-bass, 2023-01-11:
We are indeed skipping testing builds, and are totally fine including `kinematics_interface_kdl` ourselves where needed.

Thanks for clarifying!

sea-bass, 2023-01-11:
Sorry, I missed the key part of the question.

We have been using whatever is in the binaries for the 2022-11-23 sync but are in the process of moving to 2022-12-26 which is the latest sync.

Anyways, we're not building test dependencies so that package won't be there regardless and we'll make sure we import it accordingly.

destogl, 2023-01-12:
> Anyways, we're not building test dependencies so that package won't be there regardless and we'll make sure we import it accordingly.

I guess that you install deps using `rosdep`, if so, I am uncertain whether test dependencies are installed per default or not. Maybe you are using some flags to reduce footprint of your installation and therefore avoiding installation of test dependencies.

Since it seems the issue is solved, I'll close it. Fell free to reopen if needed.

sea-bass, 2023-01-12:
Now I remember! We're building from source because we're using a branch based on #423.

So for sure we are explicitly building without test depends.

Anyways, yes, closing is fine. Thank you!