ros-controls/ros2_controllers #504: [AdmittanceController] It looks like an unintentional copy-paste has happened

destogl, 2023-01-13:
Those two lines seems to be a small bug. The second line should probably be:

```
state_message_.ref_trans_base_ft.child_frame = parameters_.control.frame.id
```

https://github.com/ros-controls/ros2_controllers/blob/f4788c594026bc3ecf0d944e1fa39f3f2d40673b/admittance_controller/include/admittance_controller/admittance_rule_impl.hpp#L375-L376

rehanshah17, 2025-03-19:
@destogl  i'm interested to take this issue up and contribute. Could you help me set up / start. I have an idea of what I want to do but not too sure how to execute.

christophfroehlich, 2025-03-19:
Have a look at the [contribution guidelines on control.ros.org](https://control.ros.org/rolling/doc/contributing/contributing.html). I also added the first-issue template to the initial post. 

The solution for the fix is already given above, the question is how to test that. Have a look at the existing tests, and where the new test would fit.

rehanshah17, 2025-03-20:
I fixed _child frame_ in the test I created, and updated my pull request

closed 2025-07-25 by christophfroehlich