ros-controls/ros2_controllers #506: [AdmittanceController] Dupplication of objects in `on_init` and `on_configure` methods

destogl, 2023-01-13:
This is duplicated code in `on_init` and in `on_configure` methods, and it should be removed from `on_configure`. The objects should have lifetime of the controller, but they should be resetted. Maybe `admittance_rule_` should be created here and not in `on_init`. The reason for that is resetting parameters completely when reconfiguring controller.

https://github.com/ros-controls/ros2_controllers/blob/f4788c594026bc3ecf0d944e1fa39f3f2d40673b/admittance_controller/src/admittance_controller.cpp#L150-L160

Maverobot, 2023-09-09:
The mentioned redundancy does not seem to exist any more. Can this issue be closed? @destogl

bailaC, 2024-01-21:
@destogl  we can close this issue.

closed 2025-03-14 by christophfroehlich