ros-controls/ros2_controllers #521: Wrong frame_id in odometry message of diff_drive_controller

tverbele, 2023-02-10:
When using a namespace with the diff_drive_controller, the frame ids in the odometry message are constructed wrongly. Concretely, the namespace contains a starting "/" and is appended twice. So if I declare `odom_frame_id` as `odom` and I have `controller_namespace` being `/robot` it creates a frame id as `/robot//robot/odom` instead of the expected `robot/odom`

closed 2023-02-14 by destogl

bobbleballs, 2023-02-24:
Hi @destogl, I've also been looking into this. 
However my setup is using individual TF topics for each namespace, e.g `/robot_n/tf`
The way that diff_drive_controller does this is different to every other node I use. Others use underscores to add prefixes to frame ID's or do nothing and expect individual TF topics as described in this issue on Geometry2 /ros2/geometry2/issues/433.
Would it be possible to add an option to add namespacing, or at the very least change the `/` to a `_`?
Thanks.

bmagyar, 2023-02-26:
The version 2.17.1 in humble should have this fix. Do you experience it still? Or in rolling?

bobbleballs, 2023-02-27:
Hi yes,
Sorry I posted this in the wrong place. #482 is more related to what I'm talking about. TF frame names should be prefixed with `_` not `/` to keep in line with how a lot of other packages do it.