ros-controls/ros2_controllers #537: cannot make JointGroupVelocityController work

muttistefano, 2023-02-28:
Hi all, 
i'm using ros2_controllers on rolling to control a robot with the velocity interface.

My controllers config looks like this : 

```
forward_velocity_controller:
      type: velocity_controllers/JointGroupVelocityController

forward_velocity_controller:
  ros__parameters:
    joints:
      - shoulder_pan_joint
      - shoulder_lift_joint
      - elbow_joint
      - wrist_1_joint
      - wrist_2_joint
      - wrist_3_joint
      # interface_name: velocity
```

when I show the hardware interfaces with the CLI command, it shows:
```
command interfaces
	elbow_joint/position [available] [unclaimed]
	elbow_joint/velocity [available] [claimed]
	shoulder_lift_joint/position [available] [unclaimed]
	shoulder_lift_joint/velocity [available] [claimed]
	shoulder_pan_joint/position [available] [unclaimed]
	shoulder_pan_joint/velocity [available] [claimed]
	wrist_1_joint/position [available] [unclaimed]
	wrist_1_joint/velocity [available] [claimed]
	wrist_2_joint/position [available] [unclaimed]
	wrist_2_joint/velocity [available] [claimed]
	wrist_3_joint/position [available] [unclaimed]
	wrist_3_joint/velocity [available] [claimed]
```

The controller picks up the commands published on "/forward_velocity_controller/commands", I checked by printing them in the forwardcommandcontroller Update call.
After the controller receives the commands, they are not passed down to the hardware interface.
I also tested the hardware interface extensively, and I always receive a vector of zeros as a command.
Am I doing something wrong?

muttistefano, 2023-03-01:
Running the controller manager with level debug, shows this:

```
Deduced interface prefix 'shoulder_pan_joint' - searching for the controller with the same name.
[ros2_control_node-1] [DEBUG] [1677662455.266703502] [ControllerManager::utils]: Required command interface 'shoulder_pan_joint/velocity' with prefix 'shoulder_pan_joint' is not reference interface.
[ros2_control_node-1] [DEBUG] [1677662455.266711384] [ControllerManager::utils]: Deduced interface prefix 'shoulder_lift_joint' - searching for the controller with the same name.
[ros2_control_node-1] [DEBUG] [1677662455.266718056] [ControllerManager::utils]: Required command interface 'shoulder_lift_joint/velocity' with prefix 'shoulder_lift_joint' is not reference interface.
[ros2_control_node-1] [DEBUG] [1677662455.266725295] [ControllerManager::utils]: Deduced interface prefix 'elbow_joint' - searching for the controller with the same name.
[ros2_control_node-1] [DEBUG] [1677662455.266731897] [ControllerManager::utils]: Required command interface 'elbow_joint/velocity' with prefix 'elbow_joint' is not reference interface.
[ros2_control_node-1] [DEBUG] [1677662455.266739593] [ControllerManager::utils]: Deduced interface prefix 'wrist_1_joint' - searching for the controller with the same name.
[ros2_control_node-1] [DEBUG] [1677662455.266746390] [ControllerManager::utils]: Required command interface 'wrist_1_joint/velocity' with prefix 'wrist_1_joint' is not reference interface.
[ros2_control_node-1] [DEBUG] [1677662455.266753378] [ControllerManager::utils]: Deduced interface prefix 'wrist_2_joint' - searching for the controller with the same name.
[ros2_control_node-1] [DEBUG] [1677662455.266759739] [ControllerManager::utils]: Required command interface 'wrist_2_joint/velocity' with prefix 'wrist_2_joint' is not reference interface.
[ros2_control_node-1] [DEBUG] [1677662455.266766476] [ControllerManager::utils]: Deduced interface prefix 'wrist_3_joint' - searching for the controller with the same name.
[ros2_control_node-1] [DEBUG] [1677662455.266772893] [ControllerManager::utils]: Required command interface 'wrist_3_joint/velocity' with prefix 'wrist_3_joint' is not reference interface.
```

I cannot tell if it's related.
Thanks

destogl, 2023-03-07:
You interfaces are claimed. Therefore, they are passed. (I am quite certain because having a bug there would influence all the tests and users we have).

Which version, hardware interface are you using. What other relevant details you can provide?

daihen-hijikata, 2024-08-22:
I'm using version 2.37.1.
The commands I published with "ros2 topic pub" were often ignored.
I could see the commands with "ros2 topic echo", so I think this is JointGroupVelocityController's problem.

christophfroehlich, 2024-08-22:
Hi @daihen-hijikata. Can you reproduce your issue with [ros2_control_demos](https://control.ros.org/master/doc/ros2_control_demos/example_3/doc/userdoc.html)? If not, can you create a MWE for us to reproduce it?

christophfroehlich, 2024-08-22:
a mimimum working example, something you can share with us where we can reproduce your issue. otherwise we cannot help you..

daihen-hijikata, 2024-08-22:
I created 41joint sample robot. But my issue was not reproduced. I cannot understand that my robot often ignore the command...

christophfroehlich, 2024-08-22:
are you using the same environment (DDS vendor, network setup etc)?

daihen-hijikata, 2024-08-22:
I'm using Isaac Sim.
The control_manager run very high frequency when I set use_sim_time parameter.
similar to https://github.com/ros-controls/ros2_control/issues/859
At very high frequencies, the commands may be often ignored.

My sample is here.
https://github.com/hijimasa/isaac-ros2-control-sample/tree/control_manager_high_freq_issue
diffbot often ignore the command at very high frequencies.

daihen-hijikata, 2024-08-23:
I can reproduce my issue.
Below is the example.
[issue_sample.zip](https://github.com/user-attachments/files/16720914/issue_sample.zip)
This video was taken when the command was ignored.
https://github.com/user-attachments/assets/91c35172-17fc-46b9-bb3a-4e55f182128c

christophfroehlich, 2024-08-23:
I was not able to reproduce it with this publisher
[test_publisher.zip](https://github.com/user-attachments/files/16724947/test_publisher.zip)

I've read the linked PR above and this is for sure related to the update rate of the cm, and subsequently, the controllers. In my environment, the joint_states are sent with ~1600Hz. I guess you are using the topic_based_ros2_control with isaac sim?

christophfroehlich, 2024-08-23:
I'll close this issue here as the new issue seems to be not related at all to the initial question, and there was no update for half a year now.

closed 2024-08-23 by christophfroehlich

daihen-hijikata, 2024-08-23:
Where should I comment? Should I create new issue in ros2_controllers repo?

christophfroehlich, 2024-08-23:
IMHO that the behavior you encountered is a consequence of the update rate issue with use_sim_time and not related to the ros2_controllers repo.