ros-controls/ros2_controllers #545: Question of different behavior between Foxy and Humble

ygoumaz, 2023-03-15:
I am not really sure if my request is a bug or not, but I believe the behavior is intended and I can't find why.

In `webots_ros2_universal_robot` package, we use the `joint_trajectory_controller` package to send goal positions to multiple joints of an ABB and a UR5e robot. In Foxy, the trajectory is very smooth and it seems like multiple points are interpolated between two given trajectory points. By listening to the `/robot-name/state` topic, I can see that the publish rate is high (16 hz).

But on Humble, it seems like there is only the given trajectory points that are used, without any interpolation or any similar algorithm. In Webots, the robot has more rough movements and is really less smooth than in Foxy. When listening to the `/state` topic, the rate is very low (less than 1 hz). It seems that only the given goal positions are used. I don't know if the publishing rate has anything to do with the actual trajectory, but it really gives the impression that on Foxy there are more points taken in account.

All files, including the parameters file (shown below), are exactly the same for both distributions (Foxy and Humble):
```
/**:
  ros__parameters:
    # controller_manager
    update_rate: 20

    abb_joint_trajectory_controller:
      type: joint_trajectory_controller/JointTrajectoryController

    abb_joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

    # abb_joint_trajectory_controller
    joints:
      - A motor
      - B motor
      - C motor
      - D motor
      - E motor
      - F motor
      - finger_1_joint_1
      - finger_2_joint_1
      - finger_middle_joint_1
    command_interfaces:
      - position
    state_interfaces:
      - position
    allow_partial_joints_goal: True`
```

Is there anything that has changed in the trajectory following algorithm? Is there any parameter that can be set to get the same behavior on Humble as on Foxy?

For information [the positions](https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_universal_robot/webots_ros2_universal_robot/ur5e_controller.py) are sent from [this file](https://github.com/cyberbotics/webots_ros2/blob/master/webots_ros2_universal_robot/webots_ros2_universal_robot/follow_joint_trajectory_client.py#L93) to the `FollowJointTrajectory` client with the `send_goal_async()` function.

Thanks a lot for your time!

**To Reproduce**
Steps to reproduce the behavior:
1. Install webots_ros2
2. Start the UR example with `ros2 launch webots_ros2_universal_robot multirobot_launch.py`
3. See a different behavior on Humble and Foxy.
4. Optional: Check the /state topic rate of each robot.

**Environments**
 - OS: Ubuntu 20.04
 - Version: Foxy
 &&
- OS: Ubunut 22.04
- Version: Humble

ygoumaz, 2023-05-05:
Any idea on what change could cause the different behavior between both distributions?

bmagyar, 2023-05-05:
The algorithm hasn't changed, some handling and controller-level interfaces may have that result in what you are experiencing. That being said, smooth interpolation is the point of this controller so if it isn't happening that's an issue.

The state topic is published at a set rate, the speed of it is not indicative of issues with the controller.

How fast are commands sent to the JTC? And which interface (action/topic) is used?

ygoumaz, 2023-05-09:
The update rate is set to 50, to match the simulation timestep. We use the FollowJointTrajectory action.

The interpolation seems fine but the difference is really in the speed at which joints are getting to the goal positions. Something may have changed in gains, maximal velocity or similar. 

Here are videos to compare both behaviors. It should help understanding what is happening. We can clearly observe that movements are slower and smoother on Foxy than on Humble.

**FOXY**

https://user-images.githubusercontent.com/61198661/237043032-96ce0f98-1011-4ac4-9742-1a7b251d405f.mp4

**HUMBLE**

https://user-images.githubusercontent.com/61198661/237043047-2f21daaa-4471-4fc6-8498-ca341b37591e.mp4

ygoumaz, 2023-05-24:
Is there a way to limit the speed when using position control (Position State Interface and Position Command Interface)? It seems that the controller is trying to reach the target position as fast as possible. I have tried increasing the `time_from_start` value but all it does is add several intermediate positions for all the joints. It really feels like the interpolation is working, but the speed at the target points is not continuous (it goes down to zero).

bmagyar, 2023-05-24:
I think there was an issue with the JTC not doing interpolation properly (or at all) recently. Have you figured out more @christophfroehlich ?

christophfroehlich, 2023-05-24:
The issue I experienced is described in #574, but I think it is not related. I haven't used JTC with distros older than humble, so I can't say if something changed.

What I see that you use position command interface only, as far as I remember it interpolates with constant velocity between the given points. 

Since #520 the state publish rate was removed in rolling, in humble you can set the state publish rate, defaulting to 50Hz: https://github.com/ros-controls/ros2_controllers/blob/602b153f4cd87818204be035d85127188c319f75/joint_trajectory_controller/src/joint_trajectory_controller_parameters.yaml#L53-L60
If you don't receive it with that rate, something is broken.

If you get that running, could you plot the reference value from the state topic?

ygoumaz, 2023-05-25:
I have version 3.5.0 of JTC, and I'm currently investigating the rate of the `/state` topic. However, I'm encountering a low publish rate (1 Hz), even after adding the parameter to my YAML file (setting it to 50 Hz).

I'm wondering if this low publish rate could be the reason why the joints are attempting to move to the next position as quickly as possible? Could this be a bug in JTC, or could it be related to my ROS node that publishes the trajectory to the ActionClient?

It's worth noting that my node responsible for sending the trajectory to the `ActionClient` has a subscriber to the `/state` topic. This subscriber helps detect the first received message because the `wait_for_server()` method reports that the `joint_trajectory_controller` node is ready, even though it may require additional time to fully prepare to receive commands. I tried to remove it but it doesn't seem to fix the issue.

christophfroehlich, 2023-05-25:
the topic should be published even if no action was sent to JTC?! 

You could try to adapt the [demo JTC action client](https://github.com/ros-controls/ros2_control_demos/blob/humble/example_1/bringup/launch/test_joint_trajectory_controller.launch.py) to eliminate effects coming from your client.

I'll try to make a minimum example based on the RRbot, if it works there we can look for the differences

ygoumaz, 2023-05-26:
I haven't gone as far as updating the demo JTC client to match our project but I already came across something weird.

I have modified the [RRbot launch file](https://github.com/ros-controls/ros2_control_demos/blob/humble/example_1/bringup/launch/rrbot.launch.py) to start the `joint_trajectory_position_controller` instead of the `forward_position_controller`. 

In the original config file the `update_rate` is set to 10 and the `state_publish_rate` to 200. When I check the `/state` topic, the rate is at 10, which is probably due to the update rate limiting the state publish rate. But if I increase the `update_rate` to 400 (or any value above 200.0), the `/state` topic has a rate of 400. Shouldn't it be limited to 200 (or the value of the `state_publish_rate` parameter)?

I also tried to remove the `state_publish_rate` parameter, so that it defaults to 50 Hz. But it is always publishing at the rate given by the `update_rate` parameter. 

Is this normal behavior?

ygoumaz, 2023-05-31:
> In the original config file the `update_rate` is set to 10 and the `state_publish_rate` to 200. When I check the `/state` topic, the rate is at 10, which is probably due to the update rate limiting [...]

Note that this behavior is only present on Iron and Rolling (3.5.0). It seems to work correctly at the rate of `state_publish_rate` in Humble (version 2.21.0). Given that in our project, the state topic is published at 1Hz on Humble, Iron and Rolling, it may not be related to this.

christophfroehlich, 2023-06-05:
Starting from 3.3.0 of joint_trajectory_controller (released to iron and rolling), the `state_publish_rate` parameter was removed. I removed the remnant today also from the demos with https://github.com/ros-controls/ros2_control_demos/pull/297.

When I now launch `ros2 launch ros2_control_demo_example_1 rrbot.launch.py` with the `joint_trajectory_position_controller`, I get the correct rate
```bash
ros2 topic hz /joint_trajectory_position_controller/controller_state
average rate: 10.002
        min: 0.099s max: 0.101s std dev: 0.00053s window: 11
```
and it also changes with
```bash
controller_manager:
  ros__parameters:
    update_rate: 25  # Hz
```
to 25Hz.

With the older releases there is an issue that it publishes this topic only with the rate of the controller_manager, if `state_publish_rate` is way higher then the `update_rate` of the controller_manager. If a lower `state_publish_rate` is given, the topic is published at a fraction of the `update_rate` only.

christophfroehlich, 2023-06-06:
One suggestion: You could use the [MockComponent](https://control.ros.org/master/doc/ros2_control/hardware_interface/doc/mock_components_userdoc.html) instead of your simulator, and see if you observe the same behavior.

ygoumaz, 2023-06-07:
Thanks a lot for the references and tips, including the use of `PlotJuggler`. After hours of debugging, I finally found the origin of the issue! I have used the `ros2_control_demo` package to publish joint goal positions through the `/trajectory` topic and the issue on the state publish rate was still present. So it was definitely not a problem from our controllers publishing the goal positions on the action server. I then looked carefully at the ros2_control config files, which have the following syntax:

```xml
/**:
  ros__parameters:
    # controller_manager
    update_rate: 20

    abb_joint_trajectory_controller:
      type: joint_trajectory_controller/JointTrajectoryController

    abb_joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

    # abb_joint_trajectory_controller
    joints:
      - A motor
      - B motor
      - C motor
      - D motor
      - E motor
      - F motor
      - finger_1_joint_1
      - finger_2_joint_1
      - finger_middle_joint_1
    command_interfaces:
      - position
    state_interfaces:
      - position
    allow_partial_joints_goal: True` 
```

This is the source of the problem. For a reason, this was working in Foxy but not in later distributions. I am not at the origin of this file structure but I find it really unclear. So I converted it to:

```xml
abb/controller_manager:
  ros__parameters:
    update_rate: 50

    abb_joint_trajectory_controller:
      type: joint_trajectory_controller/JointTrajectoryController

    abb_joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster

abb/abb_joint_trajectory_controller:
  ros__parameters:
    joints:
      - A motor
      - B motor
      - C motor
      - D motor
      - E motor
      - F motor
      - finger_1_joint_1
      - finger_2_joint_1
      - finger_middle_joint_1
    command_interfaces:
      - position
    state_interfaces:
      - position

    allow_partial_joints_goal: true
```

This works perfectly and the interpolation is now completely smooth. Something in the previous syntax was completely messing the rate at which topics and nodes were running.

Thanks again for your help!

closed 2023-06-07 by ygoumaz

christophfroehlich, 2023-06-07:
Ah, I see the difference now. With the earlier yaml, you set 
```yaml
    joints:
    command_interfaces:
    state_interfaces:
    allow_partial_joints_goal: True` 
```
for all nodes in your system. But not sure how this explains your behavior, but it seems to be related with the missing namespaces.

@bmagyar or @destogl do you have an idea what the source of the problem was, and how we could be more robust here?

destogl, 2023-06-11:
If parameters are not set properly, everything is possible. That is m understanding. How knows which parameters and why controller were using.