ros-controls/ros2_controllers #548: JTC action result never returns

livanov93, 2023-03-22:
https://github.com/PickNikRobotics/moveit_studio/pull/2391

**Describe the bug**
When  `constraints.goal_time = 0.0` and there is `constraints.<joint_name>.goal` that is different than zero and can't be satisfied we always end up [here](https://github.com/ros-controls/ros2_controllers/blob/82a4e2434c8f766c22c23fc11ffb38d54404c212/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L347-L348) and the action result is never returned.

I am not sure this is desired behavior.

 
**To Reproduce**
Steps to reproduce the behavior:
1. Use https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver and change [this part](https://github.com/UniversalRobots/Universal_Robots_ROS2_Driver/blob/dfa1410653ebcb57f3522491f61a4d6344425056/ur_robot_driver/config/ur_controllers.yaml#L95-L100) into:
```
      shoulder_pan_joint: { trajectory: 0.2, goal: 0.001 }
      shoulder_lift_joint: { trajectory: 0.2, goal: 0.001 }
      elbow_joint: { trajectory: 0.2, goal: 0.001 }
      wrist_1_joint: { trajectory: 0.2, goal: 0.001 }
      wrist_2_joint: { trajectory: 0.2, goal: 0.001 }
      wrist_3_joint: { trajectory: 0.2, goal: 0.001 }
```
2. Use https://github.com/UniversalRobots/Universal_Robots_ROS2_Description and change [this part](https://github.com/UniversalRobots/Universal_Robots_ROS2_Description/blob/e53ee294b0fa86485ad6cf97b4642c7f9cc60f4c/urdf/ur.ros2_control.xacro#L34) to:
```
<param name="position_state_following_offset">-0.005</param>
```
which will make sure you never reach goal.

3.  Send action goal and see the result not returning.

mechwiz, 2023-03-29:
I could see this being the intended behavior since if you set the goal_time param to zero, that is effectively saying, wait an indefinite amount of time until the goal tolerance is met before considering the action complete. If you want to set a timeout, then the goal_time param should be set to be nonzero. that said, I could hear an argument to maybe default the goal_time param to just be some high number value if it's set to zero or not set at all just so that the action returns eventually.

olivier-stasse, 2023-03-30:
@mechwiz I agree that setting a ```goal_time``` to zero is a particular setting that should be understood as wait an indefinite amount of time. That said this is not specified in the documentation and is the default behavior as specified [here](https://control.ros.org/master/doc/ros2_controllers/joint_trajectory_controller/doc/userdoc.html). Maybe we should put it in the documentation of JTC that ```goal_time``` to zero is a particular setting.

livanov93, 2023-04-03:
I know the behavior is result of the parameter set, still it is quite terrifying thing to hear endless execution of the trajectory, even though there is always some mission oriented higher level decision maker above it.
Maybe introducing the global max duration time parameter or similar...

destogl, 2023-09-20:
Discussion on the WG Meeting on Sep, 20th – we should add a global parameter for this with a default value 0 which means indefinitely. Set this values to the "goal_timeout” filed if not set.