ros-controls/ros2_controllers #555: Admittance Controller Build Failure

swiz23, 2023-03-29:
@bmagyar @destogl 

**Describe the bug**
Admittance Controller on latest main fails to build in humble with the following error messages. Seems related to the recently merged PR #553.
```
#19 318.8 Starting >>> admittance_controller
#19 335.0 --- stderr: admittance_controller
#19 335.0 In file included from /opt/moveit2_ws/src/ros2_controllers/admittance_controller/src/admittance_controller.cpp:17:
#19 335.0 /opt/moveit2_ws/src/ros2_controllers/admittance_controller/include/admittance_controller/admittance_controller.hpp:100:37: error: ‘controller_interface::return_type admittance_controller::AdmittanceController::update_reference_from_subscribers(const rclcpp::Time&, const rclcpp::Duration&)’ marked ‘override’, but does not override
#19 335.0   100 |   controller_interface::return_type update_reference_from_subscribers(
#19 335.0       |                                     ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
#19 335.0 In file included from /opt/ros/humble/include/class_loader/class_loader/class_loader_core.hpp:57,
#19 335.0                  from /opt/ros/humble/include/class_loader/class_loader/class_loader.hpp:55,
#19 335.0                  from /opt/ros/humble/include/class_loader/class_loader/multi_library_class_loader.hpp:52,
#19 335.0                  from /opt/ros/humble/include/pluginlib/pluginlib/class_loader.hpp:38,
#19 335.0                  from /opt/moveit2_ws/src/ros2_controllers/admittance_controller/include/admittance_controller/admittance_rule.hpp:32,
#19 335.0                  from /opt/moveit2_ws/src/ros2_controllers/admittance_controller/include/admittance_controller/admittance_controller.hpp:28,
#19 335.0                  from /opt/moveit2_ws/src/ros2_controllers/admittance_controller/src/admittance_controller.cpp:17:
#19 335.0 /opt/ros/humble/include/class_loader/class_loader/meta_object.hpp: In instantiation of ‘B* class_loader::impl::MetaObject<C, B>::create() const [with C = admittance_controller::AdmittanceController; B = controller_interface::ChainableControllerInterface]’:
#19 335.0 /opt/ros/humble/include/class_loader/class_loader/meta_object.hpp:216:7:   required from here
#19 335.0 /opt/ros/humble/include/class_loader/class_loader/meta_object.hpp:218:12: error: invalid new-expression of abstract class type ‘admittance_controller::AdmittanceController’
#19 335.0   218 |     return new C;
#19 335.0       |            ^~~~~
#19 335.0 In file included from /opt/moveit2_ws/src/ros2_controllers/admittance_controller/src/admittance_controller.cpp:17:
#19 335.0 /opt/moveit2_ws/src/ros2_controllers/admittance_controller/include/admittance_controller/admittance_controller.hpp:51:7: note:   because the following virtual functions are pure within ‘admittance_controller::AdmittanceController’:
#19 335.0    51 | class AdmittanceController : public controller_interface::ChainableControllerInterface
#19 335.0       |       ^~~~~~~~~~~~~~~~~~~~
#19 335.0 In file included from /opt/moveit2_ws/src/ros2_controllers/admittance_controller/include/admittance_controller/admittance_controller.hpp:31,
#19 335.0                  from /opt/moveit2_ws/src/ros2_controllers/admittance_controller/src/admittance_controller.cpp:17:
#19 335.0 /opt/ros/humble/include/controller_interface/chainable_controller_interface.hpp:97:23: note:     ‘virtual controller_interface::return_type controller_interface::ChainableControllerInterface::update_reference_from_subscribers()’
#19 335.0    97 |   virtual return_type update_reference_from_subscribers() = 0;
#19 335.0       |                       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
#19 335.0 gmake[2]: *** [CMakeFiles/admittance_controller.dir/build.make:76: CMakeFiles/admittance_controller.dir/src/admittance_controller.cpp.o] Error 1
#19 335.0 gmake[1]: *** [CMakeFiles/Makefile2:206: CMakeFiles/admittance_controller.dir/all] Error 2
#19 335.0 gmake[1]: *** Waiting for unfinished jobs....
#19 335.0 gmake: *** [Makefile:146: all] Error 2
#19 335.0 ---
#19 335.0 Failed   <<< admittance_controller [16.2s, exited with code 2]

```
**Expected behavior**
Admittance Controller should build successfully instead of failing.

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Version: Humble
- Building ros2_controllers from source along with MoveIt

swiz23, 2023-03-29:
Just discovered that the reason this is happening is because I'm using the ros2_control debian but building ros2_controllers from source. However, it looks like this commit from yesterday (https://github.com/ros-controls/ros2_control/commit/fea2d8a398c317cd49b103ed4f716440536db5d6) in the ros2_control repo would fix this issue. So if I build ros2_control from source, this issue goes away.

closed 2023-03-29 by swiz23

bmagyar, 2023-03-29:
We are well aware :) glad you figured it out so quickly