ros-controls/ros2_controllers #568: Allow Hardware Interface To Access Params From ROS2 Controllers

swiz23, 2023-04-11:
**Is your feature request related to a problem? Please describe.**
Hi, I'm currently working on a project containing a robot and a custom hardware interface plugin. The way the robot works - you must command a robot-specific control mode to the robot before sending down commands. For example, if you want to send down positions to all joints, you send a command to the robot to get into its 'position control' mode. If you want to send down positions to some joints and velocities to other joints, you send a command to the robot to put it into a 'combo control' mode, etc...

Until now, I've coded the custom hardware interface to smartly determine what robot-specific control mode it should be sending down to the robot by way of the 'command interfaces' param that can be specified for a JTC controller . During controller activation/deactivation, the `prepare_command_mode_switch` and `perform_command_mode_switch` functions are called in the hardware interface and the desired command interfaces to start and stop are passed in as arguments. So if all the joints are only claiming the 'position' interface, then I know to command the robot to the robot-specific 'position' control mode. If a subset of the joints are claiming the 'velocity' interface, and a different subset is claiming the 'position' interface, I know I need to command the robot to be in a 'combo control' mode.

However, I've now started a part of the project where we have two different types of 'position' control modes that can be sent down to the robot. So my previous approach of determining what control mode to send to the robot based on the 'command_interfaces' breaks down here. Should the first position control mode or the second position control mode be used?

So I guess my question is...what have other people done to solve this issue? Or in general, how have other people coded their hardware interfaces to change their robot-specific control modes?

**Describe the solution you'd like**
The solution I propose is to add an optional 'data' field that can be configured within a JTC (or any other) controller in a ros2_controllers.yaml file. An example can be seen below. 
```
  trajectory_controller:
    ros__parameters:
      command_interfaces:
        - position
        - velocity
        - acceleration
      state_interfaces:
        - position
        - velocity
      joints:
        - joint_1
        - joint_2
        - joint_3
        - joint_4
        - joint_5
        - joint_6
      allow_partial_joints_goal: true
      open_loop_control: true
      constraints:
        stopped_velocity_tolerance: 0.1
        goal_time: 0.0
        joint_1: { trajectory: 0.3, goal: 0.05 }
        joint_2: { trajectory: 0.3, goal: 0.05 }
        joint_3: { trajectory: 0.3, goal: 0.05 }
        joint_4: { trajectory: 0.3, goal: 0.05 }
        joint_5: { trajectory: 0.3, goal: 0.05 }
        joint_6: { trajectory: 0.3, goal: 0.05 }
     data: position_control_mode_2
```
This `data` parameter will be sent down to the `prepare_command_mode_switch` and `perform_command_mode_switch` functions within the hardware interface (along with the 'start_interfaces' and 'stop_interfaces'). Then within the function, the hardware interface can check the value of 'data' and command the appropriate control mode accordingly. Note that 'data' doesn't have to be a 'string' type. It could be some other type that contains info that is linked to a particular controller (ex. robot-specific motor params that should be sent down upon mode switch).

Also note that you wouldn't even necessarily have to add 'data' as an argument to the `prepare_command_mode_switch` and `perform_command_mode_switch` functions. Maybe you could have a function accessible to the hardware interface that it could call to get that data anytime it wants!

**Describe alternatives you've considered**
- None

**Additional context**
The issue at https://github.com/ros-controls/ros2_control/issues/347 seems somewhat related to this question, so I'm linking to it here.

swiz23, 2023-04-14:
Thanks @gavanderhoorn, can you share a link to an example where this 'dedicated interface' approach was implemented?

gavanderhoorn, 2023-04-14:
I'm afraid not. The examples I was thinking of are all private systems.

swiz23, 2023-04-16:
Having this 'data' param though could be helpful for this PR: https://github.com/ros-planning/moveit_task_constructor/pull/355. In that PR, a feature is added to allow you to use MTC to change controllers for a particular stage. I'm not completely sure how using the 'coordination controller' would fit into that PR (how would a higher level node know when to change  the system mode?). But with the 'data' param, this issue would never surface.

destogl, 2023-05-30:
I am usually solving this by adding interfaces for each control mode, in your case that would be: `position`, `special_position`, `velocity`.

Then you always know in your HW interface which mode you have to switch to.

Here might be the limitation JTC if you want to use it in all three modes since we are fixing it to `position`, `velocity`, `acceleration` and `effort` interfaces. Nevertheless, a simple extension of it would do the job. For example, adding support also for custom position interfaces. (But are you certain that you need for custom position interface actually JTC?)

destogl, 2023-08-17:
We are using the second approach right now. For your concrete example, it might be that we need some new parameters in JTC to enable different control modes from the same type of controller, i.e. JTC.

closed 2025-05-26 by github-actions[bot]