ros-controls/ros2_controllers #571: Non time sliced trejectory controller for controlling stepper motor movements.

anand-010, 2023-04-15:
**Is your feature request related to a problem? Please describe.**
I am trying to control a robotic arm made of stepper motors to do manipulations using moveit2 on ros2 humble.
i've wrote a custom hardware interface but the data i am getting is 100Hz time sliced movement data from joint trejectory controller.
**It's very hard to control stepper motor positions smoothly from the time sliced joint data.**

**Describe the solution you'd like**
A mechanism or new controller to get non time sliced data to the hardware interface. It must have acceleration , speed or time to get to destination point. the trajectory_msgs/JointTrajectory Message seems to be all  necessary data in it.

I am new to ros2 and ros2 controls if i am said anything wrong sorry for it.

bmagyar, 2023-04-15:
So what would you like to get instead? If 100Hz is too much, would 10Hz work? Or are there any other issues you see?

anand-010, 2023-04-15:
time sliced data is not good for motors like stepper motor but it's good for servo and other type of motors.
I want to get planed trejectory positions and (time or speed or accelaration) to get the destination.
I think it's same as the trajectory_msgs/JointTrajectory Message.
so we don't need to do the time slicing inside the controller passing the raw data to the hardware interface is enough.
i don't know the architecture of the ros2 control please forgive me if any mistake.

christophfroehlich, 2025-05-21:
If you don't need interpolated data but just want to set the next target point, then use a [ForwardCommandController](https://control.ros.org/rolling/doc/ros2_controllers/forward_command_controller/doc/userdoc.html).

Closing as this issue is stale for 2 years already.

closed 2025-05-21 by christophfroehlich