ros-controls/ros2_controllers #580: Error when using Joint Trajectory Controller - Holding position due to state tolerance violation

NamanMo97, 2023-04-24:
Hello,
I am having the following error when working with JTC:

**[ros2_control_node-1] [ERROR] [1682335715.942592907] [joint_trajectory_controller]: Holding position due to state tolerance violation**

Using the following config script:

[image]

Also, is there any example on how to send desired velocity and acceleration to the controller?
Thank you!

christophfroehlich, 2023-05-12:
Hi!

The error means that the tolerance were violated while executing your trajectory. When you remove the contraints section, no tolerance should be active and the error disappears.
Without more context it is hard to tell if you set up everything properly. 

You can find an example of sending trajectories to JTC in the [demo repository](https://control.ros.org/master/doc/ros2_control_demos/example_1/doc/userdoc.html), is this what you were looking for?

christophfroehlich, 2023-07-24:
Closed due to inactivity. Feel free to reopen, giving us more context.

closed 2023-07-24 by christophfroehlich