ros-controls/ros2_controllers #584: JointTrajectoryController jiggling

andrey-pr, 2023-04-26:
Hallo. I'm trying to launch moveit servo with joint_trajectory_controller, and here is the plot of values that robot driver got on one joint:
[image]
As we can see, it returns back every published message in `/right_arm_controller/joint_trajectory` topic.
Here is the record from topic:
```
header:
  stamp:
    sec: 0
    nanosec: 0
  frame_id: leg
joint_names:
- torso_right_shoulder_top_joint
- right_shoulder_top_right_shoulder_bottom_joint
- right_shoulder_bottom_right_forearm_joint
- right_forearm_right_forearm_bottom_joint
- right_forearm_bottom_right_wrist_joint
- right_wrist_right_gripper_base_joint
points:
- positions:
  - 0.30864940821268205
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  velocities: []
  accelerations: []
  effort: []
  time_from_start:
    sec: 0
    nanosec: 150000000
---
header:
  stamp:
    sec: 0
    nanosec: 0
  frame_id: leg
joint_names:
- torso_right_shoulder_top_joint
- right_shoulder_top_right_shoulder_bottom_joint
- right_shoulder_bottom_right_forearm_joint
- right_forearm_right_forearm_bottom_joint
- right_forearm_bottom_right_wrist_joint
- right_wrist_right_gripper_base_joint
points:
- positions:
  - 0.3194334011114212
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  velocities: []
  accelerations: []
  effort: []
  time_from_start:
    sec: 0
    nanosec: 150000000
---
header:
  stamp:
    sec: 0
    nanosec: 0
  frame_id: leg
joint_names:
- torso_right_shoulder_top_joint
- right_shoulder_top_right_shoulder_bottom_joint
- right_shoulder_bottom_right_forearm_joint
- right_forearm_right_forearm_bottom_joint
- right_forearm_bottom_right_wrist_joint
- right_wrist_right_gripper_base_joint
points:
- positions:
  - 0.3302295794885404
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  velocities: []
  accelerations: []
  effort: []
  time_from_start:
    sec: 0
    nanosec: 150000000
---
header:
  stamp:
    sec: 0
    nanosec: 0
  frame_id: leg
joint_names:
- torso_right_shoulder_top_joint
- right_shoulder_top_right_shoulder_bottom_joint
- right_shoulder_bottom_right_forearm_joint
- right_forearm_right_forearm_bottom_joint
- right_forearm_bottom_right_wrist_joint
- right_wrist_right_gripper_base_joint
points:
- positions:
  - 0.3409496551449958
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  velocities: []
  accelerations: []
  effort: []
  time_from_start:
    sec: 0
    nanosec: 150000000
---
header:
  stamp:
    sec: 0
    nanosec: 0
  frame_id: leg
joint_names:
- torso_right_shoulder_top_joint
- right_shoulder_top_right_shoulder_bottom_joint
- right_shoulder_bottom_right_forearm_joint
- right_forearm_right_forearm_bottom_joint
- right_forearm_bottom_right_wrist_joint
- right_wrist_right_gripper_base_joint
points:
- positions:
  - 0.3516545102573185
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  - 0.0
  velocities: []
  accelerations: []
  effort: []
  time_from_start:
    sec: 0
    nanosec: 150000000
```
As I debugged, with every msg in topic controller sends one command with current position of joint, but not new position. Joint in emulator little bit slowly than command, but when I set it to return every command as state, it still jiggles:
[image]

**To Reproduce**
Steps to reproduce the behavior:
1. Install moveit
2. Launch an example with moveit servo

**Expected behavior**
It should work smoothly, without jiggling.

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Version: Humble, ros-humble-joint-trajectory-controller/jammy,now 2.17.2-1jammy.20230327.201927 amd64

**Additional context**
As I see by listening command topic, its not a problem of Moveit Servo, its a problem of controller.

bmagyar, 2023-04-26:
At what rate are you sending increments to the controller? What rate does the controller run?

andrey-pr, 2023-04-26:
> At what rate are you sending increments to the controller? What rate does the controller run?

Command in topic coming every 0.15 second, command from controller come to driver with 50hz (every 0.02 second)

gavanderhoorn, 2023-04-27:
@andrey-pr: you should mention https://github.com/ros-planning/moveit2/issues/2108, as that's where a lot of discussion/diagnosis has already been going on.

andrey-pr, 2023-04-27:
> @andrey-pr: you should mention https://github.com/ros-planning/moveit2/issues/2108, as that's where a lot of discussion/diagnosis has already been going on.
> 

I opened this (2108) issue, but it didn't solve problem. After hours of debugging, I decided that I should ask here, because it mostly problem of controller

gavanderhoorn, 2023-04-27:
I'm not sure why that would mean you shouldn't mention your previous issue. Quite some information in there and previous diagnosis attempts which would help the maintainers here ..

andrey-pr, 2023-04-28:
In https://github.com/ros-planning/moveit2/issues/2108 I got an tip how to solve a problem and it helped. `JointGroupPositionController` is a solution of a problem.  Now the only question I have is how to switch between controllers without stop & launch with different launch file?

closed 2023-05-10 by andrey-pr