ros-controls/ros2_controllers #642: [rqt jtc] fails to receive state since new jtc ctrl msgs

gwalck, 2023-06-02:
**Describe the bug**

following https://github.com/ros-controls/ros2_controllers/pull/557 jtc is not working anymore to read feedback, and failed at sending commands

```
KeyError: 'position'
Traceback (most recent call last):
  File "~/workspaces/ws/build/rqt_joint_trajectory_controller/rqt_joint_trajectory_controller/joint_trajectory_controller.py", line 423, in _update_cmd_cb
    pos = self._joint_pos[name]["position"]
```

**Expected behavior**
RQT JTC should get its information from the new "controller_state" topic in the new JTC message.

**Environment (please complete the following information):**
 - OS: ubuntu 22.04
 - Version master of ros2_controllers against rolling

A PR is ready. standby

bmagyar, 2023-06-02:
Ah thank you! I started looking into reproducing this a few days ago

bmagyar, 2023-06-02:
Do we also have this issue in humble?

gwalck, 2023-06-02:
@bmagyar  if jtc new control messages were ported back to humble, maybe, I did not test.

gwalck, 2023-06-02:
as far as I can see when looking at code in humble and testing a modified rrbot launch file to start a jtc instead of the forward controller on humble branch of `ros2_control_demos`
```
ros2 launch ros2_control_demo_example_1 rrbot.launch.py
```
the jtc controller of humble is publishing to both topics (state and controller_state) with a message containing both old and new structures, so the RQT JTC in humble is fine with it (well the rrbot has no joint limits I had to add some to get he RQT JTC to accept display the sliders).

destogl, 2023-06-02:
We could also update `humble` but this could also confuse people...

destogl, 2023-06-02:
Maybe to add a note that RQT plugin is using only new topic because we want people using humble to transition to the new one

gwalck, 2023-06-02:
What I understood is that humble was supposed to handle both, so we should not break RQT_JTC there. But now RQT_JTC is broken in rolling because in rolling there is only one topic accepted and one type of msg accepted.

destogl, 2023-06-08:
@mpollayil this looks like a different issue. Can you give us more informations about your setup?

What are you using from binaries, what are you using from source. This might points us to know what is happening.

mpollayil, 2023-06-08:
Hi Denis,

thanks for your prompt reply.

> @mpollayil this looks like a different issue. Can you give us more informations about your setup?

Would you like me to open a separate thread for this?
 
> What are you using from binaries, what are you using from source. This might points us to know what is happening.

We are not installing anything from source except for the [moveit_resources](https://github.com/ros-planning/moveit_resources/tree/humble) package, installed with
```
git clone https://github.com/ros-planning/moveit_resources.git
colcon build --packages-select moveit_resources_fanuc_description moveit_resources_fanuc_moveit_config
```
Our Docker file pulls [this](https://hub.docker.com/layers/ykrobotics/ros2/humble-dev/images/sha256-c8f589023c5c7c00073d3672b0accacd1975c60d9d7c6017f99480fa804cdea0?context=explore) image and then only installs the following:
```
# Install necessary ros2 packages
RUN sudo apt update -y \
    && sudo apt install -y ros-humble-controller-manager \
    && sudo apt install -y ros-humble-position-controllers \
    && sudo apt install -y ros-humble-gripper-controllers \
    && sudo apt install -y ros-humble-joint-trajectory-controller\
    && sudo apt install -y ros-humble-joint-state-broadcaster\
    && sudo apt install -y ros-humble-rqt-*   
```

Hope these information are enough. Please let me know if I should open another thread for this issue.

bmagyar, 2023-06-08:
To me those error messages seem like a convoluted way of saying "ABI incompatibility of messages". That being said I'm not really intimate with fastrtps to understand it's errors well.

destogl, 2023-06-09:
What happens if you don't start MoveIt part?

It could be that the messages are not compatible due recent changes or that libraries are not on the correct versions. Have you updated or created Docker in the last few days?

mpollayil, 2023-06-09:
@destogl A very minimal launch file like the following one would still make ros2_control node crash:
```
import os
from launch import LaunchDescription
from launch.substitutions import LaunchConfiguration
from launch_ros.actions import Node
from ament_index_python.packages import get_package_share_directory
from moveit_configs_utils import MoveItConfigsBuilder

def generate_launch_description():

    moveit_config = (
        MoveItConfigsBuilder("moveit_resources_fanuc")
        .robot_description(file_path="config/fanuc.urdf.xacro")
        .robot_description_semantic(file_path="config/fanuc.srdf")
        .trajectory_execution(file_path="config/moveit_controllers.yaml")
        .to_moveit_configs()
    )

    # ros2_control using FakeSystem as hardware
    ros2_controllers_path = os.path.join(
        get_package_share_directory("moveit_resources_fanuc_moveit_config"),
        "config",
        "ros2_controllers.yaml",
    )

    ros2_control_node = Node(
        package="controller_manager",
        executable="ros2_control_node",
        parameters=[moveit_config.robot_description, ros2_controllers_path],
        output="both",
    )

    joint_state_broadcaster_spawner = Node(
        package="controller_manager",
        executable="spawner",
        arguments=[
            "joint_state_broadcaster",
            "--controller-manager",
            "/controller_manager",
        ],
    )

    fanuc_controller_spawner = Node(
        package="controller_manager",
        executable="spawner",
        arguments=[
            "fanuc_controller",
            "--controller-manager",
            "/controller_manager",
        ],
    )

    return LaunchDescription(
        [
            ros2_control_node,
            joint_state_broadcaster_spawner,
            fanuc_controller_spawner,
        ]
    )
```
It just runs ros2_control and spawns the two controllers. So, we believe this issue might not be related to MoveIt... Can it be related to using simulating hardware with the ros2_control FakeSystem?

Regarding Docker, we did not create or update it, but we rebuilt the image from scratch a few days ago and since that day we started having these issues. With the old image everything was working perfectly.

destogl, 2023-06-09:
Does this also happens with our examples?