ros-controls/ros2_controllers #648: All Wheel Drive for Ackermann

roncapat, 2023-06-05:
It is my understanding that currently the Ackermann controller provides traction control to the rear wheels, while the front ones are passive (just steered). For my application, I have an all-wheel-drive ackermann vehicle. 

Would a patch providing all-wheel-drive option to Ackermann be welcomed here?
I would later also like to develop a 4-steerable-wheel version of the controller ("double ackermann")... would also this second one be of interest then? Most of the code may be reused.

space192, 2023-06-11:
I would love to see this feature implemented as of right now using a 4 wheel vehicule is pretty complicated when using ackermann

bmagyar, 2023-06-11:
A 4-wheel steering controller is always welcome, there was also one in ROS1 in ros_control.

Regarding the all wheel drive param in happy to incorporate it as long as it retains the current code cleanliness.

roncapat, 2023-07-07:
May I ask to have a look at #692?