ros-controls/ros2_controllers #660: ackermann_steering_controller with gazebo classic

space192, 2023-06-12:
Hi !
I was wondering if there is a command interfaces built in gazebo for ackermann_steering_controller, or do I need to code it myself ? Because right now with the following code I get the following:
`<ros2_control name="GazeboSystem" type="system">
        <hardware>
            <plugin>gazebo_ros2_control/GazeboSystem</plugin>
        </hardware>

        <joint name="fleft_steering_hinge_joint">
            <command_interface name="position" />
            <state_interface name="position"/>
        </joint>

        <joint name="fright_steering_hinge_joint">
            <command_interface name="position" />
            <state_interface name="position"/>
        </joint>

        <joint name="rleft_wheel_joint">
            <command_interface name="velocity"/>
            <state_interface name="position"/>
            <state_interface name="velocity"/>
        </joint>
        <joint name="rright_wheel_joint">
            <command_interface name="velocity"/>
            <state_interface name="position"/>
            <state_interface name="velocity"/>
        </joint>
    </ros2_control>

    <gazebo>
        <plugin name="gazebo_ros2_control" filename="libgazebo_ros2_control.so">
            <parameters>$(find arece_compet_python)/config/controllers.yaml</parameters>
        </plugin>
    </gazebo>`

[image]

bmagyar, 2023-06-12:
... and what are you expecting to get? it seems to me that you've got an ackermann vehicle and you've correctly configured and activated an ackermann controller.

 The unavailable/unclaimed interfaces are probably not needed.

space192, 2023-06-12:
the thing is when I'm trying to send a TwistStamped message on the topic my wheels just reset to 0.0 0.0 0.0 of the gazebo world 🤔  so I thought that the two hardware that were unavailable was the problem

space192, 2023-06-13:
So I just checked and when I send TwistStamped command with teleop_twist_keyboard or rqt_robot_steering, I get on the /ack_man/controller_state .inf in linear_velocity_command and -pi/2 or pi/2 wathever the command I send 🤔 do you have any idea what could be the problem ?

space192, 2023-06-14:
Hey @bmagyar sorry to bother you again, I shot a quick video (20 sec) to show the problem on my gazebo, I don't really know what's going on and verbose mode doesn't give anything :/ https://youtu.be/5mGF9LVwPs0

RayOwwen, 2023-06-16:
I had a similar problem：the car wheel in wrong place, when I load urdf model into gazebo.

[image]

I solved the problem by changing the wheel joint dnamics parameter in urdf model.

[image]

Then it work properly:

[image]

I hope this enlightens you to solve your problem.

christophfroehlich, 2023-09-29:
@space192 could you probably provide a full working example of your setup? That would be easier then to reproduce your issue.

Franklein20, 2023-12-03:
Hello 
Any update on this topic?
i got the same behaviour

christophfroehlich, 2023-12-03:
> Hello Any update on this topic? i got the same behaviour

Hi! same again, could you provide a full example of your setup?

Franklein20, 2023-12-03:
controller_manager:
  ros__parameters:
    update_rate: 1000
    use_sim_time: true

    joint_state_broadcaster:
      type: joint_state_broadcaster/JointStateBroadcaster
   

    ackermann_drive:
      type: ackermann_steering_controller/AckermannSteeringController

ackermann_drive:
  ros__parameters:

    publish_rate: 500.0  

    
    front_wheels_names: [left_wheel_steering_joint, right_wheel_steering_joint,right_back_wheel_steering_joint,left_back_wheel_steering_joint]
    rear_wheels_names: [left_wheel_joint,right_wheel_joint,wheel_back_right_joint, wheel_back_left_joint]
    #left_wheel_names: ["left_wheel_joint"]
    #right_wheel_names: ["right_wheel_joint"]
    #left_steering_name: ["left_wheel_steering_joint"]
    #right_steering_name: ["right_wheel_steering_joint"]
    front_steering: true
    velocity_rolling_window_size: 10
    position_feedback: true
    in_chained_mode: false

    wheel_separation: 0.297
    wheel_radius: 0.05

    use_stamped_vel: true

    wheel_base: 0.335    
    pivot_distance: 0.2  #0.803 

    max_steering_speed: 1.5707963267948966
    
    max_steering_angle: 0.7853981633974483

    wheel_separation_multiplier: 1.0
    left_wheel_radius_multiplier: 1.0
    right_wheel_radius_multiplier: 1.0

    publish_rate: 500.0
    odom_frame_id: odom
    base_frame_id: base_link
    pose_covariance_diagonal : [0.001, 0.001, 0.0, 0.0, 0.0, 0.01]
    twist_covariance_diagonal: [0.001, 0.0, 0.0, 0.0, 0.0, 0.01]

    open_loop: false
    enable_odom_tf: true

    cmd_vel_timeout: 0.5

    front_wheel_track: 0.297
    rear_wheel_track: 0.297
    front_wheel_radius: 0.5
    rear_wheel_radius: 0.5

joint_state_broadcaster:
  ros__parameters:
    update_rate: 1000  # Hz
    use_sim_time: true
    
    

    

here my config fil

PierRob101, 2024-05-17:
Hi anything new on this topic i'm facing the same issue
Thx in advance :)

christophfroehlich, 2024-05-18:
> [my_bot.zip](https://github.com/ros-controls/ros2_controllers/files/13540288/my_bot.zip) This zip content my package To launch the simulation , ros2 launch my_bot launch_sim.launch.py

Sorry for the late reply. You misspelled several parameters (`wheelbase` instead of `wheel_base` etc). I added a check recently, now the controller does not start if there isn't a nonzero parameter set. Please check the [docs](https://control.ros.org/rolling/doc/ros2_controllers/ackermann_steering_controller/doc/userdoc.html) or the yaml file defining the parameters.

Furthermore, the ackermann_steering_controller assumes that the two wheels in front are steerable, and the two in the back are traction wheels. It seems that your robot has a steering joint an all wheels. The following configuration
```
    front_wheels_names: [left_wheel_steering_joint, right_wheel_steering_joint]
    rear_wheels_names: [wheel_back_left_joint, wheel_back_right_joint, ]
```
kind of works, but fails in simulation because the back wheels are not actuated.

@space192 I see also some misspelled parameters. Have you managed to fix your configuration in the meantime?
@PierRob101 does this information help you as well?

PierRob101, 2024-05-21:
@christophfroehlich  Thank you for your Reply :) I checked my config and URDF files for the things you mentioned, but it seems my problem is caused by something else if you want to help me fuhrer investigating you can have a look at my Setup here:
[rover_description.zip](https://github.com/ros-controls/ros2_controllers/files/15387631/rover_description.zip)

christophfroehlich, 2024-05-26:
I'll close this here because it seems that every issue was caused by wrong parameters (the check got released to all distros). 

If you still encounter issues, please see my comment above about tuning gazebo physics. If you have problems with that, post a question into [RSE](https://robotics.stackexchange.com/questions) or the gz repositories.
There are some bugs with the steering library still, follow #1118 and #1150.

closed 2024-05-26 by christophfroehlich

PierRob101, 2024-05-29:
Thank you :) i'm feeling dumb for not recognizing the s's in my config i managed to get everything working now :+1: