ros-controls/ros2_controllers #673: [JTC

tingelst, 2023-06-16:
**Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - OS: [e.g. iOS]
 - Version [e.g. Foxy]
- Anything that may be unusual about your environment

**Additional context**
Add any other context about the problem here, especially include any modifications to any ros2_controllers that relate to this issue.

tingelst, 2023-06-16:
Seems like I am not the only one having this issue: https://github.com/frankaemika/franka_ros2/pull/10#issuecomment-1594526860

tingelst, 2023-06-19:
Thank you, @JafarAbdi!  I can confirm that I got the expected behavior, that is, the robot motion smooth, when I checked out the parent commit: https://github.com/ros-controls/ros2_controllers/commit/ce0aeb29e8b42a030248c95a6ee3ec0e99c5e80b. I will continue testing tomorrow.

tingelst, 2023-06-20:
@JafarAbdi, I have found the source of the jerky/wobbly motion. The bug was introduced in https://github.com/ros-controls/ros2_controllers/commit/ce0aeb29e8b42a030248c95a6ee3ec0e99c5e80b as you expected. The velocity error term is not computed when using an effort command interface, only for velocity command interfaces. This can be fixed by changing [this line](https://github.com/ros-controls/ros2_controllers/blob/6c979091a7e6477963226765a23ecadd33a70d3a/joint_trajectory_controller/src/joint_trajectory_controller.cpp#LL136C9-L136C38) from
```cpp
    if (has_velocity_state_interface_ && has_velocity_command_interface_)
```
to
```cpp
    if (has_velocity_state_interface_ && (has_velocity_command_interface_ || has_effort_command_interface_))
```

I'll submit at PR for this.

I also have a problem with non-zero effort commands commanded after execution as mentioned in issue https://github.com/ros-controls/ros2_controllers/issues/671#issue-1757801230, but I'll take that discussion there.

JafarAbdi, 2023-07-20:
@tingelst This should be closed since the bug is fixed in the main branch, right?

tingelst, 2023-07-20:
@JafarAbdi, yes indeed. Thank you for the help in tracking this down.

closed 2023-07-20 by tingelst