ros-controls/ros2_controllers #675: Steering controllers (ackermann, bicycle, tricycle) respecting joint limits

ARK3r, 2023-06-16:
Setting up a demo for the Ackermann controller I keep noticing that when I send turn commands the steering keeps turning until it reaches unreasonable angles and then the model starts to jerk around uncontrollably after the steering angles get out of proportion to each other:

[Screencast from 06-16-2023 12:08:30 PM.webm](https://github.com/ros-controls/ros2_controllers/assets/22005547/0e64489a-563d-4d15-987c-d67f2ba7243e)

I think the 3 of steering controllers could use a steering limit parameter in their yaml configuration files to stop giving commands farther than the limits.

This feature is mostly needed for the software's understanding of what the robot is doing and calculating odometry, considering in the real world the robot would just physically be unable to turn more than the limits but if the software doesn't know this, the software's understanding of the state of the robot could start drifting from the reality of the physical state of the robot.

christophfroehlich, 2023-10-10:
Are you sure that this isnt the same behavior as we found out with https://github.com/ros-controls/ros2_control_demos/pull/316? (velocity vs position command if).

As you describe that the real robot is limiting the steering angles, wouldn't it make sense to implement that in your hardware component? Then all should be fine if you are not in open_loop mode. I agree that the limits should also be considered inside the steering lib (especially for open_loop mode).

ARK3r, 2023-10-10:
I think the controller should not overshoot the realistic limits of steering anyway, even if open_loop is true, otherwise the robot description gets to unusable states.

I do think however this issue will become more clear as we figure out how to configure velocity for rear drive and position for steering.

cschindlbeck, 2024-08-27:
We have an ackermann-type robot with steering angle joint limits +/- 45 deg, this would be highly useful

christophfroehlich, 2024-08-27:
> We have an ackermann-type robot with steering angle joint limits +/- 45 deg, this would be highly useful

This means that your commanded twist is not feasible. Shouldn't this be considered by the path controller on top of the steering controllers?

cschindlbeck, 2024-08-27:
> > We have an ackermann-type robot with steering angle joint limits +/- 45 deg, this would be highly useful
> 
> This means that your commanded twist is not feasible. Shouldn't this be considered by the path controller on top of the steering controllers?

so far, we have only connected a joypad controller which does not have any limits imposed, we will set the limit on the motor controller and in future we will use nav2 for path creation. 

diff drive has limits on linear x and angular z, wouldn't this make sense for the steering controllers as well?

christophfroehlich, 2025-05-21:
This is now handled by the RM, added recently with https://github.com/ros-controls/ros2_control/pull/2047

closed 2025-05-21 by christophfroehlich