ros-controls/ros2_controllers #701: [JTC] Is acceleration-only command interface setup useful?

christophfroehlich, 2023-07-14:
The validator does not allow acceleration-only command interfaces:
https://github.com/ros-controls/ros2_controllers/blob/6851df137cf2118d1a7166921d34eb759c225526/joint_trajectory_controller/include/joint_trajectory_controller/validate_jtc_parameters.hpp#L33-L36

I'm not sure if this would make sense, i.e., if there is any robot type with acceleration-only interface, but I think JTC could work with it.

- Should we allow it?
- If not -> I'll clarify that in the [docs](https://github.com/ros-controls/ros2_controllers/blob/master/joint_trajectory_controller/doc/userdoc.rst#hardware-interface-type-1), because this is not clearly described there.

destogl, 2023-09-09:
You are right. If we have effort, then acceleration is not necessary. People using direct control of joint moments will use rather effort than acceleration.

So I am arguing to disable that.

christophfroehlich, 2023-09-09:
Thanks for the info, I'll write an update for the docs for clarification.

PR #770 (2023-09-16): [Docs] Improve interface description of JTC
This should improve the description of supported combinations of state and command interfaces and closes #701.
closed 2023-09-16 by bmagyar