ros-controls/ros2_controllers #744: JointGroupPositionController should respect velocity limits

DLu, 2023-08-14:
**Is your feature request related to a problem? Please describe.**
As described [here](https://robotics.stackexchange.com/q/25100/35046) I'm trying to port old urdf tutorials. 

In ROS 1, you could limit the speed that the position controller moved at by specifying a `<limit>` tag within the `<joint>` tag. See [this part of the tutorial](http://wiki.ros.org/urdf/Tutorials/Using%20a%20URDF%20in%20Gazebo#Joint_Control)

I'm trying a similar thing in ROS 2, but the `<limit>` tag has no effect, still snapping immediately. [Check out this branch and this section of the partially rewritten tutorial](https://github.com/ros/urdf_sim_tutorial/tree/ros2_devel#joint-control)

**Describe the solution you'd like**
The functionality that existed in ROS 1

**Describe alternatives you've considered**
Switching back to ROS 1 forever.

bmagyar, 2023-08-15:
I agree it should.

Due to early insecurities about param handling with ROS2 we plowed ahead
without the controllers having access to the URDF.

Now there's a PR that fixes that so it opens up the possibility of
respecting joint limits once again automagically (some controllers already
do by having duplicated limits info in their configs).

bmagyar, 2023-08-15:
Please rate your experience 1-10 with our customer success team professionals today!!

christophfroehlich, 2025-05-21:
This is done by the resource manager, recently added with https://github.com/ros-controls/ros2_control/pull/2047