ros-controls/ros2_controllers #745: Odd QOS for Diff Drive TF Publication

DLu, 2023-08-14:
**Describe the bug**
Is there a particular reason why [the TF is published directly using a publisher](https://github.com/ros-controls/ros2_controllers/blob/83e415c0f00512398df3c3a508f738fc76a7ecda/diff_drive_controller/src/diff_drive_controller.cpp#L443-L444) rather than TF Broadcaster? Publishing it with a custom publisher / custom QOS settings does not play nicely with other systems. 

**To Reproduce**
1. `ros2 launch gazebo_ros2_control_demos diff_drive.launch.py`
2. Attempt to visualize TF in rviz2

**Expected behavior**
Full transform from /odom to all other links. 

**Actual behavior**
Full transforms not published, most frequently failing to publish odom -> base_link/chassis (See https://github.com/ros-controls/gazebo_ros2_control/pull/224)

**Environment (please complete the following information):**
 - OS: Ubuntu 20.04
 - Version Foxy

bmagyar, 2023-08-22:
I believe when this was ported, the broadcaster wasn't a thing yet.
Please feel free to submit a PR

DLu, 2023-11-07:
Okay, so the **real** issue is that raw publisher is passed to the `realtime_tools::RealtimePublisher`, and I don't think there's a `RealtimeTFBroadcaster`. Should the QOS just get fixed, or is there something else todo with realtime?

christophfroehlich, 2023-11-07:
Maybe it's worth it to implement a `RealtimeTFBroadcaster` based on the [realtime_publisher](https://github.com/ros-controls/realtime_tools/blob/master/include/realtime_tools/realtime_publisher.h)?

bmagyar, 2023-11-07:
Afaik the broadcaster/listener are tfbuffer+publisher/subscriber. We could also just grab the publisher from inside the broadcaster class if that's a possibility in API