ros-controls/ros2_controllers #755: 1000hz JTC issue

bmagyar, 2023-08-23:
Hi @gwalck :

I am facing a robot jerky problem when using JCT with Moveit in ROS2 Foxy. But, the problem did not appear in ROS1 Noetic. So, I also dug deeper in sample methods between ros_control and ros2_control. As you mentioned above, ros2_control  implementation samples at current time. So, I set the loop rate of ros2_control_node at frequency 1000hz  and print the sample time out like this:

[image]

When I tried to plan and excute with Moveit, the sample time was shown in my terminal.
[image]

It is clear that, although the trajectory carried out by spline is smooth, the different sample interpolation may cause a jerky robot. And, I am afraid that the rate limit in [ros2_control_node ](https://github.com/ros-controls/ros2_control/blob/33e77b8d28cc5a29780b0a13f669934c564a5d8c/controller_manager/src/ros2_control_node.cpp#L86)will not work due to that the sample interpolation is longer than 1ms in my situation.

Do I miss something causing my jerky robot? Have you met the same problem like me?  If it possible, could you please show me  some solutions to handle this jerky problem?

platform: ubuntu 20.04
ROS Version: Foxy

_Originally posted by @zengxuhe in https://github.com/ros-controls/ros2_controllers/issues/697#issuecomment-1683449512_

gwalck, 2023-08-24:
@bmagyar  did you forget to move my post ?

destogl, 2023-09-09:
@zengxuhe your output times look quite irregular. Are you using RT Kernel? Without it you will not get 1000 Hz stable.

christophfroehlich, 2025-05-21:
> [@bmagyar](https://github.com/bmagyar) did you forget to move my post ?

it seems so, @gwalck do you remember and could repost your answer here? I can't access it any more in the other thread.

gwalck, 2025-05-21:
Hi, I apparently used only the webinterface to answer and I did not receive email notifications of my own contribution. The only way to recover my exact answer is to ask @zengxuhe to check in his email notifications around August 23rd 2023 and copy it back here. I don't remember what I answered but I looked again, and don't think the problem was related to the sampling being at "now" instead of "now+dt", but could be due to the realtime loop as mentioned by @destogl