ros-controls/ros2_controllers #759: set_hold_position results into not exactly reaching desired target

firesurfer, 2023-08-29:
**Describe the bug**

In #558  (included in rolled out version 3.14.0)  `set_hold_position` was added which commands the current position as the target after sampling the last point of the trajectory (https://github.com/ros-controls/ros2_controllers/blob/102c1cf698acd80b10c00e19a901a82e066358d4/joint_trajectory_controller/src/joint_trajectory_controller.cpp).

This results into not reaching the target exactly. (In our case a UR16e arm).
As far as I see there is no way to disable or tune this behavior

**To Reproduce**

Take some robot arm. Move rather fast from A to B. Check if the desired target has been reached within a tolerance of for example 0.001rad 

**Expected behavior**

In case the checks (e.g. arm is within goal tolerance) were successful set the last point of the trajectory as target

**Environment (please complete the following information):**
- iron
- ros2control : 3.14.0
-

firesurfer, 2023-09-04:
@christophfroehlich  If I understand the code correctly the goal tolerances are just there for checking if the goal has been reached after sampling the last point of the trajectory. If the arm takes just a tiny bit of time to precisely reach the target this check would fail. 

I worked around this in the previous version of the JTC by implementing our own check after the action has been executed successfully, but this only works under the precondition that the JTC actually sets the last point of the trajectory as desired target.

christophfroehlich, 2023-09-04:
this is not 100% correct, it checks for he duration of the goal_time parameter, not only once. 
this means, if you set a nonzero goal_time it should act as you desire

firesurfer, 2023-09-06:
I did set the goal_time. Seems to work now. It was just confusing because there was a change in behavior compared to the previous version.

Nevertheless I think 

> We could add an option for using the desired states instead of the measured ones.

would be rather nice to have.

PR #842 (2023-12-01): [JTC] Continue with last trajectory-point on success
As discussed in #811: Continue with last trajectory point on success, instead of hold-position from current state.

Closes #811 and  closes #759.
closed 2023-12-01 by bmagyar