ros-controls/ros2_controllers #762: steering controller library (ackermann steering) does not publish odom TF

guidout, 2023-09-03:
**Describe the bug**
I see the following topics:
/ackermann_steering_controller/controller_state
/ackermann_steering_controller/odometry
/ackermann_steering_controller/reference
/ackermann_steering_controller/tf_odometry
/ackermann_steering_controller/transition_event
but the odom transform is not being published.

**Environment:**
 - OS: Ubuntu 22.04
 - Version: Humble

Corund060, 2023-09-20:
/ackermann_steering_controller/tf_odometry is odom tf, remap to /tf

guidout, 2023-09-20:
I tried that but it doesn't seem to apply. Maybe I have an error in the launch file?
Here is how I'm launching the controller in my launch file:
```
ackermann_steering_controller_spawner = Node(
       package="controller_manager",
       executable="spawner",
       parameters=[{'use_sim_time': use_sim_time}],
       arguments=["ackermann_steering_controller"],
       remappings=[('ackermann_steering_controller/tf_odometry','tf')],
       condition= IfCondition(LaunchConfiguration('use_ackermann_controller'))
    )
```

Corund060, 2023-09-20:
You are remapping in the wrong place. You have to do it for ros2_control not for the spawner.

guidout, 2023-09-21:
yes! that worked. Thank you so much.
Now that works for simulation, for the real robot I guess I have to had the remappings in ros2_control node like below, right?
```
controller_manager = Node(
        package="controller_manager",
        executable="ros2_control_node",
        parameters=[{'robot_description': robot_description},
                    controls_params_file,
                    controls_rear_wheels_params_file,
                    {'use_sim_time': use_sim_time}],
        remappings=[('ackermann_steering_controller/tf_odometry','tf')],
        condition= UnlessCondition(use_sim_time)
    )
```

Corund060, 2023-09-22:
This should work. Haven't tested it myself though.

christophfroehlich, 2023-09-29:
Yes this should be fine, see [this example](https://github.com/ros-controls/ros2_control_demos/pull/316) here.

closed 2023-09-29 by christophfroehlich