ros-controls/ros2_controllers #764: [gripper] Add tests for effort interface

christophfroehlich, 2023-09-06:
We should definitely add tests for that to improve the gripper_controller. @chama1176 Could you have a look on that? I suggest copying the parameterized tests from joint_trajectory_controller, so that we run all tests with position and effort interface.

_Originally posted by @christophfroehlich in https://github.com/ros-controls/ros2_controllers/pull/756#pullrequestreview-1613036898_

christophfroehlich, 2023-09-06:
@chama1176 Could you have a look on that?

chama1176, 2023-09-06:
@christophfroehlich 
Thank you for your comment.
I will check it out.

christophfroehlich, 2023-09-07:
Great. A good starting point is the [linked tutorial](https://www.sandordargo.com/blog/2019/04/24/parameterized-testing-with-gtest) from here
https://github.com/ros-controls/ros2_controllers/blob/5b86e3c848a0f1498978e24fe5b1b48566c2d14e/joint_trajectory_controller/test/test_trajectory_controller_utils.hpp#L540-L555
and
https://github.com/ros-controls/ros2_controllers/blob/5b86e3c848a0f1498978e24fe5b1b48566c2d14e/joint_trajectory_controller/test/test_trajectory_controller.cpp#L1631-L1640

PR #769 (2023-09-08): [gripper] Add test for effort gripper controller
Add test for effort gripper controller.

Fix #764 

GripperController uses template parameters, so I used type-parameterized tests.
https://github.com/google/googletest/blob/main/docs/advanced.md#type-parameterized-tests
https://en.cppreference.com/w/cpp/types/integral_constant
closed 2023-09-08 by bmagyar