ros-controls/ros2_controllers #783: Fatal Error: Missing 'rclcpp/event_handler.hpp' during Build of master

Andreasgdp, 2023-09-28:
**Describe the bug**
When attempting to build the ros-controls repository using the `colcon build --symlink-install` command, a fatal error is encountered. The error message indicates that the file `rclcpp/event_handler.hpp` is missing or cannot be located.

**To Reproduce**
Steps to reproduce the behavior:
1. Clone repo
2. Build repo with `colcon build --symlink-install`
3. See that you get include error with `"rclcpp/event_handler.hpp"`

**Expected behavior**
The build process should proceed without any errors or missing file references.

**Screenshots**
If applicable, add screenshots to help explain your problem.
[image]

**Environment:**
 - OS: Ubuntu 22.04
 - Version Humble

gavanderhoorn, 2023-09-28:
Can you check you're building the `humble` branch?

Only `master` tries to `#include` that file.

christophfroehlich, 2023-09-28:
> Can you check you're building the `humble` branch?
> 
> Only `master` tries to `#include` that file.

correct, there was an [upstream change](https://github.com/ros-controls/ros2_controllers/pull/610) of the header files from iron on, and the rolling-version (from master branch) won't build on a humble distro.

Andreasgdp, 2023-09-28:
That was the issue, tanks

closed 2023-09-28 by Andreasgdp