ros-controls/ros2_controllers #787: No apt package for range-sensor-broadcaster

flochre, 2023-10-01:
**Describe the bug**
In apt list there are no package ros-humble-range-sensor-broadcaster

**To Reproduce**
Steps to reproduce the behavior:
1. $ apt list | grep range-sensor

**Expected behavior**
A clear and concise description of what you expected to happen.

An apt should be created for the package

**Screenshots**
If applicable, add screenshots to help explain your problem.

[image]

If I do the same for another package then it is appearing with the updated date of September 20th

$ apt list | grep diff-drive-controller
[image]

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Version Humble

**Additional context**
Add any other context about the problem here, especially include any modifications to any ros2_controllers that relate to this issue.

The package is new from PR #725 and #766 maybe just a small step has been forgotten because it is there for iron and rolling

Thanks in advance

christophfroehlich, 2023-10-02:
There was no release of the humble branch since the backport. 
I'm sure that @bmagyar will release it for the next sync to humble.

bmagyar, 2023-10-03:
Thanks for raising this guys, I forgot to push a release.

bmagyar, 2023-10-03:
This needs to be merged + some grace period needs to elapse before they sync it to the upstream apt. With ROSCon around the corner I'd not hold my breath. Expect it to be available upstream Humble in 3 weeks.

https://github.com/ros/rosdistro/pull/38683

closed 2023-10-03 by bmagyar