ros-controls/ros2_controllers #811: Send feedback for action only once

destogl, 2023-11-01:
In the below part of the code (and other parts where we set action result) should be executed only once. One possibility to do this would be by using `rt_has_pending_goal_` flag.

This should resolve the issue of #759 in a simpler way, without adding a new parameter which increases complexity. (@christophfroehlich FYI)

https://github.com/ros-controls/ros2_controllers/blob/1dac309979fb9392dc12feb6b46555582fed6fc6/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L335C13-L341C56

christophfroehlich, 2023-11-01:
Sorry I'm not sure if I understand you right, what is the problem with the linked code? maybe I get it wrong, but it should already be called only once because with the next update cycle the old goal is not active any more?
The conclusion of the linked PR was to add an option to hold the last command instead of the last measured state?!

christophfroehlich, 2023-11-15:
Summary of todays meeting:
We want JTC to command the last point of the trajectory instead of set_hold_position. If a user sends nonzero-velocity points in the end (with `allow_nonzero_velocity_at_trajectory_end=true`), it should keep moving.

PR #842 (2023-12-01): [JTC] Continue with last trajectory-point on success
As discussed in #811: Continue with last trajectory point on success, instead of hold-position from current state.

Closes #811 and  closes #759.