ros-controls/ros2_controllers #816: [JTC] Need for `start_with_holding` option?

christophfroehlich, 2023-11-06:
During discussions with @saikishor about #811 the question came up, if `start_with_holding=false` behavior is useful at all?

The code could be simplified if the default `start_with_holding=true` is the only option.

Some context: the `set_hold_position` feature from #558 was motivated from the following PRs/issues:

- #531
- https://github.com/ros-controls/ros2_controllers/issues/514
- https://github.com/ros-controls/ros2_controllers/issues/683
- https://github.com/ros-controls/ros2_controllers/issues/469

I can't remember now anyone asking for `set_hold_position=false` functionality. This was the behavior before #558 and added for legacy reasons.

Is there a use-case for this or should we delete this option and don't confuse people with it? See https://github.com/ros-controls/ros2_controllers/pull/801#discussion_r1383250510 
As it wasn't backported to humble yet, I think there is not a big use of the current behavior and no one would miss it.

padhupradheep, 2023-11-06:
Okay, in our case, we have a linear axis, where after we do the homing, we move the axis 10 cm. However, after that we give the control to JTC, which actually moves the robot down to the 0.0. Of course, we have set the initial position to 0.1 m with the params. 

In my opinion, I'd probably keep it, however I highly recommend to rename the variable.

christophfroehlich, 2023-11-06:
> Okay, in our case, we have a linear axis, where after we do the homing, we move the axis 10 cm just with motor commands (had to configure like that for some special reason). However, after that we give the control to JTC, which actually moves the robot down to the 0.0. Of course, we have set the initial position to 0.1 m with the params.

Why do you need JTC to be active during homing? Couldn't you activate it after homing, then you will have the correct initial commands?
And there is no initial position parameter for JTC, which one do you mean?

christophfroehlich, 2023-11-06:
> In my opinion, I'd probably keep it, however I highly recommend to rename the variable.

What would be a more descriptive parameter name for you (it is not a variable in this sense)?

saikishor, 2023-11-06:
> Okay, in our case, we have a linear axis, where after we do the homing, we move the axis 10 cm just with motor commands (had to configure like that for some special reason). However, after that we [...]

I agree with @christophfroehlich that you could activate the JTC after the homing. This should be a part of your system configuration, where you activate JTC after performing homing. Moreover, how are you able to command the motor separately, having the JTC active? The command interfaces are already taken by the JTC  right? 

Same question, which initial position parameters do you mean?

padhupradheep, 2023-11-06:
Sorry, if my comment was confusing. 

I meant we do the homing, move the axis 10 cm after homing and then give the control to the joint trajectory controller. To be clear once again, homing and moving the axis 10 cm does not require JTC. Those are just mere position commands given directly to the motor controllers after successful initialisation of HW interface.

> (it is not a variable in this sense)?

Again sorry, was in a hurry.. 

> which initial position parameters do you mean?

I meant this line.

https://github.com/neobotix/neo_mpo_700-2/blob/a75e773cd010d6d5ca87793d0411abc97d47d613/robot_model/mpo_700/mpo_700.urdf#L378

maybe I misunderstood the usage of this particular parameter?

> What would be a more descriptive parameter name for you (it is not a variable in this sense)?

good question.. I’ll get back on that..

christophfroehlich, 2023-11-07:
> I meant we do the homing, move the axis 10 cm after homing and then give the control to the joint trajectory controller. To be clear once again, homing and moving the axis 10 cm does not require JTC. Those are just mere position commands given directly to the motor controllers after successful initialisation of HW interface. With 'set_holding' param set to true, the JTC moves the axis down to the 0 position. However setting this particular param false keeps the actual position we wanted it to be.

But why do you set JTC active during homing? You could do

- Load, configure, and activate your hardware component
- Do the homing manually
- Load, configure, and activate JTC: it takes the position at time of activation.

> > which initial position parameters do you mean?
> 
> I meant this line.
> 
> https://github.com/neobotix/neo_mpo_700-2/blob/a75e773cd010d6d5ca87793d0411abc97d47d613/robot_model/mpo_700/mpo_700.urdf#L378
> 
> maybe I misunderstood the usage of this particular parameter?

The URDF is not used at all by any ros2_controller (as of now): This has no effect on JTC.

padhupradheep, 2023-11-07:
> Load, configure, and activate JTC: it takes the position at time of activation.

Thanks for this point, I was actually activating JTC along with the Hardware component. I get your question now. 

Probably should have read the documentation better

destogl, 2023-11-15:
I think we should remove this option. IMHO, I expect from controller to send command immediately after it is activated. And if a robot should not move then send the same command over and over again.

closed 2023-12-01 by bmagyar