ros-controls/ros2_controllers #861: [Iron] - FTS Broadcaster crashes after latest Iron sync

firesurfer, 2023-11-27:
**Describe the bug**

After the latest ros2 iron sync the `force_torque_sensor_broadcaster` crashes with a Segmentation fault when loading the controller at startup. When I manually spawn the controller at a later point everything seems to work fine.

```
[ros2_control_node-6] #13   Source "../sysdeps/unix/sysv/linux/x86_64/clone3.S", line 81, in __clone3 [0x7f3e17126a3f]
[ros2_control_node-6] #12   Source "./nptl/pthread_create.c", line 442, in start_thread [0x7f3e17094ac2]
[ros2_control_node-6] #11   Object "/usr/lib/x86_64-linux-gnu/libstdc++.so.6.0.30", at 0x7f3e174dc252, in 
[ros2_control_node-6] #10   Object "/opt/ros/iron/lib/librclcpp.so", at 0x7f3e1790e9d9, in rclcpp::executors::MultiThreadedExecutor::run(unsigned long)
[ros2_control_node-6] #9    Object "/opt/ros/iron/lib/librclcpp.so", at 0x7f3e178ff125, in rclcpp::Executor::execute_any_executable(rclcpp::AnyExecutable&)
[ros2_control_node-6] #8    Object "/opt/ros/iron/lib/librclcpp.so", at 0x7f3e179056dc, in rclcpp::Executor::execute_service(std::shared_ptr<rclcpp::ServiceBase>)
[ros2_control_node-6] #7    Object "/opt/ros/iron/lib/librclcpp.so", at 0x7f3e179cd579, in 
[ros2_control_node-6] #6    Object "/opt/ros/iron/lib/libcontroller_manager.so", at 0x7f3e17d262fa, in 
[ros2_control_node-6] #5    Object "/opt/ros/iron/lib/libcontroller_manager.so", at 0x7f3e17cfd7fb, in std::_Function_handler<void (std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Response_<std::allocator<void> > >), std::_Bind<void (controller_manager::ControllerManager::*(controller_manager::ControllerManager*, std::_Placeholder<1>, std::_Placeholder<2>))(std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Response_<std::allocator<void> > >)> >::_M_invoke(std::_Any_data const&, std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Request_<std::allocator<void> > >&&, std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Response_<std::allocator<void> > >&&)
[ros2_control_node-6] #4    Object "/opt/ros/iron/lib/libcontroller_manager.so", at 0x7f3e17cd2fe6, in controller_manager::ControllerManager::configure_controller_service_cb(std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Request_<std::allocator<void> > >, std::shared_ptr<controller_manager_msgs::srv::ConfigureController_Response_<std::allocator<void> > >)
[ros2_control_node-6] #3    Object "/opt/ros/iron/lib/libcontroller_manager.so", at 0x7f3e17cc84e4, in controller_manager::ControllerManager::configure_controller(std::__cxx11::basic_string<char, std::char_traits<char>, std::allocator<char> > const&)
[ros2_control_node-6] #2    Object "/opt/ros/iron/lib/libcontroller_manager.so", at 0x7f3e17cf3ccd, in 
[ros2_control_node-6] #1    Object "/opt/ros/iron/lib/libcontroller_manager.so", at 0x7f3e17cdd469, in controller_manager::ControllerManager::controller_sorting(controller_manager::ControllerSpec const&, controller_manager::ControllerSpec const&, std::vector<controller_manager::ControllerSpec, std::allocator<controller_manager::ControllerSpec> > const&)
[ros2_control_node-6] #0    Object "/opt/ros/iron/lib/libforce_torque_sensor_broadcaster.so", at 0x7f3e0c27d455, in force_torque_sensor_broadcaster::ForceTorqueSensorBroadcaster::state_interface_configuration() const
[ros2_control_node-6] Segmentation fault (Address not mapped to object [(nil)])
```

**To Reproduce**

Have a multi robot setup with the Universal Robots Ros2 Driver and spawn a force torque sensor at startup.

Excerpt from my launch file
```
    fts_spawner_top = Node(
        package="controller_manager",
        executable="spawner",
        arguments=["force_torque_sensor_broadcaster_top",
                   "--controller-manager", "/controller_manager"],
    )
    fts_spawner_bottom = Node(
        package="controller_manager",
        executable="spawner",
        arguments=["force_torque_sensor_broadcaster_bottom",
                   "--controller-manager", "/controller_manager"],
    )

```

controllers.yaml
```
force_torque_sensor_broadcaster_top:
  ros__parameters:
    sensor_name: ur_top/tcp_fts_sensor
    state_interface_names:
      - force.x
      - force.y
      - force.z
      - torque.x
      - torque.y
      - torque.z
    frame_id: ur_top/tool0
    topic_name: ft_data

force_torque_sensor_broadcaster_bottom:
  ros__parameters:
    sensor_name: ur_bot/tcp_fts_sensor
    state_interface_names:
      - force.x
      - force.y
      - force.z
      - torque.x
      - torque.y
      - torque.z
    frame_id: ur_bot/tool0
    topic_name: ft_data
```

**Expected behavior**

Should not crash

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - OS:  Ubuntu 22.04
 - Version: ros iron  
 - force_torque_sensor_broadcaster: 3.17.0
- Multiarm setup

saikishor, 2023-11-27:
Hello @firesurfer!

Thank you for opening the issue. Could you please clone the fixes in this PR: https://github.com/ros-controls/ros2_control/pull/1180 and let us know if this solves your issues?. Can you also tell us how many controllers are loaded in the controller_manager before loading these broadcasters?

Thank you,

firesurfer, 2023-11-27:
@saikishor
Thanks for your answer!.
Actually at the point where this happens it is the first controller that is loaded. 
So I am not sure if this PR will actually help.

saikishor, 2023-11-27:
@firesurfer 

Thank you for the information. Can you please confirm the ros2_control version that is currently installed in your machine?. The version: [3.21.1](https://github.com/ros-controls/ros2_control/commit/fbdc894dfa48965d1567db9e6a92b12f2d5ed99e) should fix the issue you might be facing. Your issue seems to be something similar to : https://github.com/ros-controls/ros2_control/issues/1164

Thank you,

Best Regards,
Sai

firesurfer, 2023-11-28:
@saikishor I have version 3.21.0 of the controller_interface. That's the version currently in the repository. 
I think this issue matches perfectly to what Felix wrote in: https://github.com/ros-controls/ros2_control/issues/1164

So until the next package sync the only solution will be building ros2_control manually I guess?

saikishor, 2023-11-28:
Hello @firesurfer!

Yes, the next sync should fix your issue. The only solution until then is to build it locally.

Thank you,

Best Regards,
Sai

firesurfer, 2023-11-28:
Thanks a lot for your help. I will close this issue now as it is basically solved!

closed 2023-11-28 by firesurfer