ros-controls/ros2_controllers #882: Dynamically tuning [joint_trajectory_controller] PID gain

JinbaoXiong, 2023-12-05:
In ros1, the change of P parameters of effort_controller/JTC in runtime can be use to change the force of the joint. But in ros2, I have tried to change the P parameters using the /set_parameters service provided by the JTC in runtime, but the forece still remain same.
I just wander, when the JTC in ros2 can have the ability to accept the change of the PID gains in runtime like ros1?

closed 2023-12-05 by JinbaoXiong

christophfroehlich, 2023-12-05:
Yes, in iron and rolling. 
Iron might not be released yet, you have to build it from source.
For humble you can use the the branch from [this PR](https://github.com/ros-controls/ros2_controllers/pull/801) and build it from source.

christophfroehlich, 2023-12-05:
Sorry, for iron it was already released with 3.18 too.

JinbaoXiong, 2023-12-05:
thanks a lot!!!!