ros-controls/ros2_controllers #891: rqt_joint_trajectory_controller: Use urdf_parser

christophfroehlich, 2023-12-07:
Maybe we can use the upstream [urdf_parser_py](https://github.com/ros/urdf_parser_py/tree/ros2/)? The mentioned issue got fixed years ago..

https://github.com/ros-controls/ros2_controllers/blob/925a690fcb469cb705ec67185cd67c6f1481dd61/rqt_joint_trajectory_controller/rqt_joint_trajectory_controller/joint_limits_urdf.py#L17-L20

bailaC, 2024-01-21:
Hello @christophfroehlich , are we going to do this work? If yes, I can give it a try.

christophfroehlich, 2024-01-21:
no one is working on that afaik. Thanks for picking this.
I just had the idea when we were fixing a bug. Please check if it really makes sense/supports all we need before you start coding!

christophfroehlich, 2025-07-26:
@bailaC any progress on this?