ros-controls/ros2_controllers #894: rqt_joint_trajectory_controller refurbishment

christophfroehlich, 2023-12-09:
## Background

rqt_jtc should be fully functional since #890 again, but it could see some refurbishment.

I copied the todos from [rqt_jtc](https://github.com/ros-controls/ros2_controllers/blob/master/rqt_joint_trajectory_controller/rqt_joint_trajectory_controller/joint_trajectory_controller.py) to this issue to make it more visible. 

@Noel215 wrote this when porting this to ROS 2 with #356, I leave this uncommented here and open for discussion.

> - Better UI support for continuous joints (see DoubleEditor TODO)
> - Can we load controller joints faster?, it's currently pretty slow
> - If URDF is reloaded, allow to reset the whole plugin?
> - Allow to configure:
>   - URDF location
>   - Command publishing and state update frequency
>   - Controller manager and jtc monitor frequency
>   - Min trajectory duration
> - Fail gracefully when the URDF or some other requisite is not set
> - Could users confuse the enable/disable button with controller start/stop
>   (in a controller manager sense)?
> - Better decoupling between model and view
> - Tab order is wrong. For the record, this did not work:
>   QWidget.setTabOrder(self._widget.controller_group,
>                       self._widget.joint_group)
>   QWidget.setTabOrder(self._widget.joint_group,
>                       self._widget.speed
> 

## Instructions
Hi, this is a `good-first-issue` issue. This means we've worked to make it more legible to people who either **haven't contributed to our codebase before, or even folks who haven't contributed to open source before**.

We're interested in helping you take the first step, and can answer questions and help you out along the way. Note that we're especially interested in contributions from underrepresented groups!

We know that creating a pull request is the biggest barrier for new contributors. This issue is for you 💝

If you have contributed before, **consider leaving this PR for someone new**, and looking through our general [bug](https://github.com/ros-controls/ros2_controllers/labels/bug) issues. Thanks!

### 🤔 What you will need to know.

Nothing. This issue is meant to welcome you to Open Source :) We are happy to walk you through the process.

### 📋 Step by Step

- [ ] 🙋 **Claim this issue**: Comment below. If someone else has claimed it, ask if they've opened a pull request already and if they're stuck -- maybe you can help them solve a problem or move it along!

- [ ] 🗄️ **Create a local workspace** for making your changes and testing [following these instructions](https://docs.ros.org/en/rolling/Tutorials/Workspace/Creating-A-Workspace.html), for Step3 use "Download Source Code" section with [these instructions](https://control.ros.org/master/doc/getting_started/getting_started.html#building-from-source).

- [ ] 🍴 **Fork the repository** using the handy button at the top of the repository page and **clone** it into `~/ws_ros2_control/src/ros-controls/ros2_controllers`, [here is a guide that you can follow](https://guides.github.com/activities/forking/) (You will have to remove or empty the existing `ros2_controllers` folder before cloning your own fork)

- [ ] **Checkout a new branch** using `git checkout -b <branch_name>`

- [ ] 🤖 **Apply `pre-commit`** auto formatting, by running `pip3 install pre-commit` and running `pre-commit install` in the ros2_control repo.

- [ ] 💾 **Commit and Push** your changes

- [ ] 🔀 **Start a Pull Request** to request to merge your code into `master`. There are two ways that you can start a pull request:
1. If you are not familiar with GitHub or how to create a pull request, [here is a guide you can follow](https://guides.github.com/activities/hello-world/) on how GitHub works.

- [ ] 🏁 **Done** Ask in comments for a review :)

### Is someone else already working on this?

🔗- We encourage contributors to link to the original issue in their pull request so all users can easily see if someone's already started on it.

👥- **If someone seems stuck, offer them some help!**

### 🤔❓ Questions?

Don’t hesitate to ask questions or to get help if you feel like you are getting stuck. For example leave a comment below!
Furthermore, you find helpful resources here:
* [ROS2 Control Contribution Guide](https://control.ros.org/master/doc/contributing/contributing.html)
* [ROS2 Tutorials](https://docs.ros.org/en/rolling/Tutorials.html)
* [ROS Answers](https://answers.ros.org/questions/)

**Good luck with your first issue!**

Ashray4, 2025-04-03:
Hey is this still open i would like to take this up.Cheers

Ashray4, 2025-11-01:
Hey, Sorry for a realy late update , i have so far managed this section 
Allow to configure:

    URDF location
    Command publishing and state update frequency
    Controller manager and jtc monitor frequency
    Min trajectory duration
i am working on the "DoubleEditor TODO" section. i should be able to finish everything by end of next week