ros-controls/ros2_controllers #896: Trajectory replacement strategy for JTC

pac48, 2023-12-10:
The [Trajectory Replacement](https://control.ros.org/master/doc/ros2_controllers/joint_trajectory_controller/doc/trajectory.html) docs indicate that the ROS 1 trajectory replacement strategy is implemented in ROS 2. It looks like 
[this](https://github.com/ros-controls/ros2_controllers/issues/574) PR aimed to address this issue already. However, there is a comment on in [joint_trajectory_controller.cpp](https://github.com/ros-controls/ros2_controllers/blob/999eae5491a79f9734f8d5e4e18ffea3e7a04e2c/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L1099) that indicates the ROS  1 trajectory replacement technique is not implemented. The ROS 2 JTC appears to completely replace the current trajectory with the new one, rather than merging them together as described in the docs. From ROS 1, I believe the replacement logic happens in the 'command' topic callback. Here is a [reference](https://github.com/ros-controls/ros_controllers/blob/67b0e721e245fb744e33891508104a133bd2f59a/joint_trajectory_controller/include/joint_trajectory_controller/init_joint_trajectory.h#L203). 

As far as fixing this issue, the logic of merging trajectories can likely be added [here](https://github.com/ros-controls/ros2_controllers/blob/999eae5491a79f9734f8d5e4e18ffea3e7a04e2c/joint_trajectory_controller/src/trajectory.cpp#L55).

christophfroehlich, 2023-12-10:
Thanks for writing this down. I'm already aware of this, summarized:

- Preemption of an active goal discards the old trajectory, and depending of time_from_start/header time stamp it has a smooth transition or not (as I discussed in #574)
- Sending a topic while having an active goal overrides the current trajectory without cancelling the goal (the comment you linked), with the same transition behavior as above. I'm not sure if we should cancel the old goal in this case?

What really is missing and wrong in the current docs: If the start of the new trajectory (independent of topic/action interface) is in the future (header time stamp, start_now), JTC now does not use the points from the old trajectory from now until start_now of the new one, it just fades the current state to the first point of the new trajectory.

christophfroehlich, 2023-12-10:
I think this issue is a duplicate of #84 btw.

pac48, 2023-12-10:
@christophfroehlich It does look like the same issue. Should we consolidate the discussion on that issue? If so, then I can close this one.

christophfroehlich, 2023-12-10:
Yes, maybe let's keep the discussion in a single issue

closed 2023-12-10 by pac48