ros-controls/ros2_controllers #930: Improve Steering Controller Tests

christophfroehlich, 2023-12-14:
> Yes, you're right, there are a lot of things open to test -- but right now I just wanted to provide a quick fix for the Ackermann odometry, only rudimentary tested. Maybe we should open a new issue, tagged "help wanted" for proper testing of the steering lib -- starting with tests for the integrator functions, etc. But maybe first we should check for a redesign cf. #692

ok, I'll create a new issue then.

> BTW, I just added this file 'test_steering_odometry.cpp', with some hard-coded tests. Is that the way to go, or do we want to have a more advanced testing style?

Should the expectations of the FW kinematics be true for all the different kinematic configurations? Then we could change this to a parameterized test as we have with JTC
https://github.com/ros-controls/ros2_controllers/blob/1d0d7531ef8d24ea52e761a872b8bf277652dd33/joint_trajectory_controller/test/test_trajectory_controller_utils.hpp#L695-L710

https://github.com/ros-controls/ros2_controllers/blob/1d0d7531ef8d24ea52e761a872b8bf277652dd33/joint_trajectory_controller/test/test_trajectory_controller.cpp#L1840-L1849

_Originally posted by @christophfroehlich in https://github.com/ros-controls/ros2_controllers/issues/921#issuecomment-1851853293_

Vedant87, 2025-03-26:
Hi, I would like to contribute.

Dylan-Gallagher, 2026-08-17:
Hi, I would like to help with this. I audited the current master tests and would propose a focused, test-only change in test_steering_odometry.cpp rather than reviving the earlier skip-heavy parameterization:

- one table-driven fixture whose rows execute the same assertions for bicycle, tricycle, and Ackermann inputs
- closed-form pose expectations from a nonzero initial pose for straight/RK2 and curved/exact integration
- coverage for the correct position-feedback overloads, distinct steering/traction tracks, reset, small-dt behavior, and invalid configuration
- before/after coverage plus a temporary mutation check to show the new oracles catch broken geometry

I see maintainer PR #1695 is active and changes both the tricycle API and this test file. Would you prefer that I wait and rebase the full test patch after #1695, or start with integration plus Ackermann/bicycle coverage that avoids the tricycle paths? I will keep the contribution DCO-compliant and disclose AI assistance according to the project policy.

christophfroehlich, 2026-08-17:
Welcome! I suggest that you split this in several PRs to be able to review this.