ros-controls/ros2_controllers #931: Improve Steering Controllers Documentation

christophfroehlich, 2023-12-14:
> In general, I'd appreciate more documentation of the kinematics in the docs for all controllers here. Is there any publicly available reference/book for these mobile robot kinematics we could refer to? 
I'd happy to help with writing the docs, and maybe writing proper tests for the implementations. But I'm not "fluent" with these type of kinematics, I'd need a textbook for that.

_Originally posted by @christophfroehlich in https://github.com/ros-controls/ros2_controllers/issues/878#issuecomment-1837235341_

christophfroehlich, 2023-12-14:
For example, some simple sketches showing the system inputs of the different steering geometries. Pictures with appropriate licenses could be used and adapted, e.g., from [wikipedia](https://en.wikipedia.org/wiki/Ackermann_steering_geometry#/media/File:Ackermann_turning.svg). This might also be related with #692

bmagyar, 2023-12-14:
I'd suggest relying on Wikipedia as much as possible. The core equations are usually well explained but figures may be lacking.

christophfroehlich, 2023-12-14:
I haven't found anything appropriate to Ackermann steering for example :man_shrugging:

ARK3r, 2023-12-14:
This is a resource that I found useful: https://msl.cs.uiuc.edu/planning/node657.html

christophfroehlich, 2023-12-29:
Some literature review

- From [above](https://msl.cs.uiuc.edu/planning/node657.html) or as [PDF version](https://lavalle.pl/planning/booka4.pdf) (**Tri-Cylce, Simple Car**)
- @franzrammerstorfer mentioned [Lynch, Kevin M., and Frank C. Park. Modern Robotics: Mechanics, Planning, and Control 1st ed. Cambridge, MA: Cambridge University Press, 2017](https://hades.mech.northwestern.edu/index.php/Modern_Robotics), which has a [PDF preprint](https://hades.mech.northwestern.edu/images/2/25/MR-v2.pdf) available. (**Car-Like Robot**)
- [Siciliano et.al  Robotics: Modelling, Planning and Control](https://link.springer.com/book/10.1007/978-1-84628-642-1) (**Bicycle**)
- [Ribeiro, Lima - KINEMATICS MODELS OF MOBILE ROBOTS](http://users.isr.ist.utl.pt/~mir/cadeiras/robmovel/Kinematics.pdf) (**Tricycle**)

They all handle the kinematics of car-like robots in state-space form, but none is discussing odometry etc. of the different geometries @ARK3r has pointed to [here](https://github.com/ros-controls/ros2_controllers/issues/692#issuecomment-1749748573). I guess we have to write that on our own :nerd_face:

closed 2025-01-01 by christophfroehlich