ros-controls/ros2_controllers #966: [JTC][Humble] Failed to configure controller because some parameters are read-only

Noel215, 2024-01-10:
Hi, 

After the last sync of humble the joint_trajectory_controller is not working anymore on TIAGo simulation. 

For each of the JTC we get this output: 
```
[spawner-8] [ERROR] [1704887660.109800268] [spawner_arm_controller]: Set parameter joints failed: parameter 'joints' cannot be set because it is read-only
[spawner-8] Set parameter command_interfaces failed: parameter 'command_interfaces' cannot be set because it is read-only
[spawner-8] Set parameter state_interfaces failed: parameter 'state_interfaces' cannot be set because it is read-only
```

I've been investigating a bit and it seems related to https://github.com/ros-controls/ros2_controllers/pull/771. Ping @christophfroehlich

**To Reproduce**
Starting from a fresh `humble` docker:
```
source /opt/ros/humble/setup.bash
sudo apt update
sudo apt install ros-humble-tiago-simulation
ros2 run tiago_gazebo tiago_gazebo.launch.py
```

christophfroehlich, 2024-01-10:
Thanks for reporting. I currently use JTC with humble successfully, so is is not broken in general. I think we still have the problem with generate_parameter_library and init/config method @saikishor?

saikishor, 2024-01-11:
This is a resume after the direct conversation with @christophfroehlich 

This will happen if you don't have the parameters are the creation of the node, as the read-only parameters are expected to be set before.

When using spawners, we have the load_controller first instantiating the LifeCycleNode and then when the GPL library trying to declare and get the parameters, as they don't exist you see this error in the first place. 

We have to come up with some kind of mechanism, so this kind of issue won't happen again. Moreover, I agree with @christophfroehlich that the parameters that were marked as read-only make sense as they are not supposed to change dynamically. However, the current situation is due to the limitation of ROS2.

christophfroehlich, 2024-01-11:
This means that we currently can't use any read_only parameters at all.
We started using it because we don't have a mechanism for JTC to dynamically change the degrees-of-freedom etc. (and we don't want to do that).

christophfroehlich, 2024-01-11:
Some thoughts on how to fix that:

- change the load_controller service so that the lifecycle node is created at the time of service call and not at the time of spawning the controller_manager. Is this possible in general @saikishor?
- Try to change upstream ros2 parameter to enable parameter override of lifecycle nodes (maybe @saikishor can share his experience in doing so)
- add parameter-set callbacks to the controllers to prevent changing of the should-be-read-only parameters (verbose, and not very nice from the controller-code-side)