ros-controls/ros2_controllers #1072: Changing the upstream controller's goal status in chained controllers

samialperen, 2024-03-04:
Imagine a chained controller: JTC (upstream) + admittance (downstream)

Is there a way that the downstream chained controller (admittance controller) can set upstream goal (JTC's goal) as successful? Otherwise, admittance might want to do nothing, however that is not going to affect JTC's goal which will eventually timeout that is undesired.

christophfroehlich, 2024-07-18:
No, this is not possible at the moment. The action itself is not shared between controllers.

christophfroehlich, 2024-07-19:
Per default it is even not possible to share any information between controller. However, it now should be possible to use state interfaces by the chained controller to push information from the downstream to the upstream controller.

closed 2024-07-19 by christophfroehlich