ros-controls/ros2_controllers #1109: Allow option to compute odometry from unpowered wheels

wittenator, 2024-04-28:
**Is your feature request related to a problem? Please describe.**
In the current ackermann controller the odometry is computed from the steering angle, the kinematic setup and the rotation of the traction wheels. Since the traction wheels have slip in contrast to unpowered wheels (which may be quite influential for car-like robots with high inertia), the odometry gets negatively affected by this. Maybe this also affects tickets like #937 since the default gazebo setup often results in high slip driving situations in my experiments.

**Describe the solution you'd like**
It would be great to have an option to specify the non-traction wheels in the config of the ackermann controller and use them for the odometry computation.

**Describe alternatives you've considered**
It would probably be possible to write a function on top of the ackermann controller that corrects for these issues by computing the slip, but that seems like an unnecessarily complex setup.

christophfroehlich, 2024-04-29:
Hi! I was thinking a bit in this direction, to make the steering_library more configurable with setups as you describe. It stopped with https://github.com/ros-controls/ros2_controllers/pull/954 waiting for reviews (you want to proofread it and leave a review there)?
But I certainly won't have time before the Jazzy release to work on this again.

Juliaj, 2025-04-09:
@christophfroehlich , we discussed this during today's PMC meeting. I'm interested in working on this. Could you assign it to me?

Juliaj, 2025-04-13:
> the default gazebo setup often results in high slip driving situations in my experiments.

@wittenator, do you still observe this issue with recent Jazzy builds? If so, would you mind attaching the URDF (or SDF) file of your robot to this issue?

I followed the instructions from https://github.com/ros-controls/gz_ros2_control/blob/jazzy/doc/index.rst and tested the demo for ackermann_drive_controller, but was unable to reproduce the issue.

wittenator, 2025-04-17:
I also wasn't able to reproduce it. Seems like the issue that I linked to actually fixed that. This issue is still relevant though for real high-inertia robots. I think that this can be reproduced in Gazebo by upping the allowed torque on the wheels and giving the robot higher inertia and a slippery ground.

closed 2025-04-19 by christophfroehlich

Juliaj, 2025-04-23:
> [@Juliaj](https://github.com/Juliaj) there are several other issues discussing a refactoring of the library, e.g., [#692](https://github.com/ros-controls/ros2_controllers/issues/692)
> Start by leaving a review at [#1166](https://github.com/ros-controls/ros2_controllers/pull/1166) ;)

Will do, will do 😃 . Some how, this type of mentions don't always show up on my git notifications.

Juliaj, 2025-04-23:
> I think that this can be reproduced in Gazebo by upping the allowed torque on the wheels and giving the robot higher inertia and a slippery ground

@wittenator, I got a few questions if you can look into them whenever you have the chance.

This scenario could be beneficial for real robots so I spent sometime to look it. Would you take a look at this [SDF file](https://github.com/Juliaj/ros2_control_demos_ext/blob/main/example_1/description/gazebo/test_ackermann_drive.sdf) on the torque and ground configuration ?  I also created a test script to drive the bot with relatively high speed with turns, you can find it [here](https://github.com/Juliaj/ros2_control_demos_ext/blob/main/example_1/bringup/launch/demo_test_helper.py). The instructions to run it is at [here](https://github.com/Juliaj/ros2_control_demos_ext/blob/main/example_1/doc/userdoc.rst). With these, I haven't able to observe slippage in Gazebo simulation.