ros-controls/ros2_controllers #1138: JTC computes wrong trajectory_end_time during validate_trajectory_msg

KolinGuo, 2024-05-21:
**Describe the bug**
joint_trajectory_controller computes wrong `trajectory_end_time` during `validate_trajectory_msg`.

In [JointTrajectoryController::validate_trajectory_msg()](https://github.com/ros-controls/ros2_controllers/blob/master/joint_trajectory_controller/src/joint_trajectory_controller.cpp#L1377-L1392) method, the `trajectory_end_time` is computed by adding the `time_from_start` of all trajectory points to `trajectory_start_time`. 

However, if I understand correctly, the `time_from_start` values are all referenced from the `header.stamp` and are not accumulated (delta time between two trajectory points). 

**Expected behavior**
Correctly compute `trajectory_end_time` as
`trajectory_end_time = trajectory_start_time + trajectory.points.back().time_from_start`

**Environment (please complete the following information):**
 - OS: Ubuntu 22.04
 - Version: Humble & Rolling

christophfroehlich, 2024-05-22:
There is already a PR fixing this, see #1090

KolinGuo, 2024-05-22:
Ahh I see, didn't notice the PR previously. Thanks for the pointer!

christophfroehlich, 2024-05-22:
No problem at all, thanks for reporting ;) But you could have a look at the PR and give your approval if this works for you (Files changed -> Review Changes)

closed 2024-05-22 by bmagyar