ros-controls/ros2_controllers #1156: joint_trajectory_controller tests fails with rmw_zenoh

CihatAltiparmak, 2024-05-29:
**Describe the bug**
The tests of ros2_control fails with rmw_zenoh. For more, please look at [this issue](https://github.com/ros2/rmw_zenoh/issues/186)

**To Reproduce**
Steps to reproduce the behavior:

Build [given dockerfile](https://gist.github.com/CihatAltiparmak/2a660fc25af9aba23c54aa2e3882be38#file-ros2_control_testing-dockerfile) using below commands

Shell 1:
```
docker build -f ros2_control_testing.Dockerfile -t ros2_control_testing .
docker run -it ros2_control_testing
# Inside docker container, just run
cd ws
source /opt/ros/iron/setup.bash
source install/setup.bash
export RMW_IMPLEMENTATION=rmw_zenoh_cpp
gdb ./build/joint_trajectory_controller/test_trajectory_actions
## Inside gdb
run
```

Just for running rmw_zenohd
Shell 2:
```
docker exec -it <running_ros2_container_name> bash
source /opt/ros/iron/setup.bash
source install/setup.bash
export RMW_IMPLEMENTATION=rmw_zenoh_cpp
ros2 run rmw_zenoh_cpp rmw_zenohd
```

**Expected behavior**
ros2_control tests must be passed with rmw_zenoh middleware.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Environment (please complete the following information):**
 - OS: docker image
 - Version iron

**Additional context**

I just wanted to inform this community. I also validated that the tests of ros2_controllers and ros2_control pass with rmw_fastrtps. This error is probably related to rmw_zenoh and looks like the bug of rmw_zenoh. In addition, rmw_zenoh is the experimental middleware.

christophfroehlich, 2025-03-28:
@CihatAltiparmak was this fixed upstream?

CihatAltiparmak, 2025-04-02:
Hello, unfortunatelly, I have no time to give a try again and feedback you. It was committed a lot of major changes like switching zenoh-c to zenoh-cpp since this issue was opened. Maybe I can give a try in my free time which I cannot make it for a while. Closing out for now on.

closed 2025-04-02 by CihatAltiparmak